zynthian/zynthian-ui #238: MIDI Timecode causes MIDI input lamp to be permanently on

riban-bw, 2020-04-17:
**Describe the bug**
Feeding a stream of MIDI timecode into the MIDI input results in the MIDI input indicator in the topstrap being permanently on.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect external MIDI device producing MIDI timecode to MIDI input
2. Play some notes on a MIDI keyboard

**Expected behaviour**
MIDI indicator in topstrap indicates the notes being played by keyboard.

**Actual behaviour**
MIDI indicator in topstrap permanently indicates MIDI timecode being received.

**Setup:**
 HARDWARE
Raspberry Pi 4 Model B Rev 1.2
Soundcard: HifiBerry DAC+ light
Display: WaveShare 7 HDMI+USB
Wiring: I2C_HWC
GPIO Expander: Not detected

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-04-09

SOFTWARE
zyncoder: master (3513de2)
zynthian-ui: master (77fd40b)
zynthian-sys: master (ccad6f8)
zynthian-data: master (9cccef5)
zynthian-webconf: master (ab294aa)

**Additional context**
It may be advantageous to indicate MTC with another indicator, e.g. underline the MIDI indicator space and show 'm' when other MIDI data is received, e.g.
[image] [image] [image]

riban-bw, 2020-04-19:
The underscore looks jagged and wrong:
[image]

jofemodo, 2020-05-19:
It should be solved now. Please confirm!

jofemodo, 2020-06-10:
Is it solved now, right?

riban-bw, 2020-06-11:
No! I just checked out master and see the following behaviour:
- Play a MIDI note: 'm' shows in status - expected behaviour
- Start clock: 'm' with underscore shows in status - unexpected behaviour

Whilst the MIDI clock is running, even without any other MIDI messages occurring, there is a permanent underscored 'm' displayed.

Webconf MIDI monitor shows only SYS CLOCK messages.

jofemodo, 2020-06-11:
Try now ...

riban-bw, 2020-06-12:
Thanks! Working now.

commit 28a74f69 (2020-06-11): Fix #238: use a separated tkinter item (line) for MIDI clock status.
closed 2020-06-12 by riban-bw