zynthian/zynthian-ui #240: MIDI CC and Note Toggle

smiths73v3, 2020-05-01:
**Is your feature request related to a problem? Please describe.**
Momentary CC buttons and pads on my controller should be able to toggle CC or notes on and off.

**Describe the solution you'd like**
I would like the ability to MIDI map CC and Note events to act as toggles.

**Describe alternatives you've considered**
Some higher end MIDI controllers provide this functionality, but I have a mix of controllers and some do not provide this. It would be more consistent to have this functionality in the Zynthian.

**Additional context**
Eg. Toggle Channel 0 CC 7 (Volume) between 32 and 96 each time Channel 0 CC 21 is activated:
`MAP CH#0 CC#21 => CH#0 TC#7 VAL#32,96`

Eg. Toggle Channel 0 CC 20 between 0 and 127 each time Channel 10 Note On 43 is activated:
`MAP CH#10 NON#43 => CH#0 TC#20 VAL#0,127`

Eg. Toggle Note Channel 1 Note 43 between 0(off) and 82 each time Channel 1 CC 70 is activated:
`MAP CH#1 CC#70 => TN#43 VAL#0,82`

Further expansion could include range on input:
Eg. 4 ranges for CC 21, mapped to 4 controller range:
`MAP CH#0 CC#21 VAL#0,32     => CH#0 CC#70 VAL#32,64`
`MAP CH#0 CC#21 VAL#33,64   => CH#0 CC#71 VAL#32,64`
`MAP CH#0 CC#21 VAL#65,96   => CH#0 CC#72 VAL#32,64`
`MAP CH#0 CC#21 VAL#97,127 => CH#0 CC#73 VAL#32,64`

Eg. make a rotary switch from a CC knob encoder:
`MAP CH#0 TC#21 VAL#0,32     => CH#0 TC#70 VAL#0,127`
`MAP CH#0 TC#21 VAL#33,64   => CH#0 TC#71 VAL#0,127`
`MAP CH#0 TC#21 VAL#65,96   => CH#0 TC#72 VAL#0,64`
`MAP CH#0 TC#21 VAL#97,127 => CH#0 TC#73 VAL#64,0`

jofemodo, 2020-05-01:
It's implemented, although in a simplified way. Less is more ;-)
Update and try these rules:

TOGGLE NON#60
TOGGLE CC#23

Or by channel:

TOGGLE CH#15 NON#60
TOGGLE CH#15 CC#23

Regards!

smiths73v3, 2020-05-03:
In zynmidirouter.c line 884, it looks like toggle will only send a val of zero.
 `884: zynmidi_send_ccontrol_change(event_chan, event_num, 0);`
I was expecting the toggle to keep "state" information First "val>0", send val of 127. Second press, send val of 0.
Here's a crude hack of what I am thinking......

			if ((zmip->flags & FLAG_ZMIP_TOGGLE) && event_val>0) {
				if (event_type==CTRL_CHANGE && midi_filter.cc_toggle[event_chan][event_num]>0) {
			               if (midi_filter.cc_toggle[event_chan][event_num]==1) {
					   zynmidi_send_ccontrol_change(event_chan, event_num, 127);
                                       midi_filter.cc_toggle[event_chan][event_num]==2
                                   }else
                                   {
					   zynmidi_send_ccontrol_change(event_chan, event_num, 0);
                                       midi_filter.cc_toggle[event_chan][event_num]==1
                                    }
				}
				else (similar fixes for note events)
`
`

mheidt, 2020-05-04:
Do I have to filter the eventtype listbox? Right now, all eventtypes are possible

jofemodo, 2020-06-24:
Hi @smiths73v3 !
Have you tried the new MIDI-chain with some of the MIDI-tool plugins? It should solve your use-case ...
Anyway, i would try to fix the feature on zynthian router ...

Regards,

smiths73v3, 2020-07-09:
Yeah, the new MIDI-chain plugins look promising, but seem a bit "alpha" at the moment. I'll dig around these a bit and see if I can offer any useful improvements.

jofemodo, 2020-07-10:
It's not alpha at all. It's working quite well ...

smiths73v3, 2020-07-12:
The concept is working well, but I found that the MIDI remap plugin, and
the Unisono plugin did not work as expected. Also, the MIDI CC range plugin
seems to limit all CC's to a range, rather than selecting some CCs for
range limit. Note toggle, and single CC remap were both working well.

jofemodo, 2020-11-20:
OK! I have removed the "TOGLE" feature from the router. The implementation is not useful as i didn't understand well your requirements. I think it could be better implemented as a MIDI-FX plugin, improving some of the currently included, or adding a new one. Programming this kind of plugins is not difficult at all ...

closed 2020-11-20 by jofemodo