zynthian/zynthian-ui #241: Storing snapsho name with the voice name as default

dcoredump, 2020-05-03:
**Is your feature request related to a problem? Please describe.**
When saving a snapshot only a number is displayed until I edit this snapshot in webconf to giv it  a name. Mostly I am using the default voice name.

**Describe the solution you'd like**
Why not using a string combination of Engine_Voicename? Would make things much easier and clearer when storing a bunch of sounds in a session.

**Describe alternatives you've considered**
No name and only a number is not an alternative...

**Additional context**
...

jofemodo, 2020-05-07:
@dcoredump  ... please could you confirm that i'm understanding your request correctly?

dcoredump, 2020-05-07:
Sorry for answering late,I've had much to work and had some sad things to deal with over the past few days.

So: I think it's better to use the name of the first layer instead of nothing, so: YES :-)

When I am at the rehearsal-rootm and tweaking around with Zynthian, I am often storing some snapshots. When not renaming with webconf and trying to figure out after a week which one was the one I currently having in my brain,  a name is better than onyl the number.

Thanks!!!

jofemodo, 2020-05-07:
OK! I will implement it ASAP ;-)

BTW, have you tried ZS3 (subsnapshots)? I'm very interested on getting feedback about them ...

jofemodo, 2020-05-07:
Done! Pleae, update and test ...

closed 2020-05-07 by jofemodo