zynthian/zynthian-ui #243: MIDI effect layer

Baggypants, 2020-05-10:
We have a Synth layer and an Audio Effect Layer. I think we need a MIDI effect layer for adding in arpeggiator lv2's and the like., This would claim a MIDI channel and clone it's output to other channels where synths would be.

jofemodo, 2020-05-11:
Yes. I'm thinking about this subject from a time ago, and there is some open proposals regarding this, like:

https://github.com/zynthian/zynthian-ui/issues/155

So, perhaps it's a good moment to step forward and implement it ... ;-)

jofemodo, 2020-05-31:
It's almost implemented ...

a56c8e9
0144ffb

Those of you wanting to test this new feature, simply select the “midi-chain” branch for UI’s repository. You can use the webconf for that.

Ahhh!, don’t forget to “enable” some MIDI tool plugin from the webconf’s LV2-plugins panel :wink:

jofemodo, 2020-05-31:
BTW, i enjoyed a lot chaining  MIDI-chord + Arpeggiator

commit fd1add6b (2020-06-01): Fix #243: Fix bugs on MIDI-chain implementation. See https://discourse.zynthian.org/t/midi-fx-chain/4240/3
commit a5fcdfed (2020-06-02): Fix #243: Fix removing MIDI-FX layer.
commit 07311efe (2020-06-07): Fix #243: Fix MIDI routing & MIDI network autoconnection.
closed 2020-06-09 by jofemodo