zynthian/zynthian-ui #245: reboot: pianoteq instrument with max. volume regardless of previous setting

hal80, 2020-05-16:
**Describe the bug**
Pianoteq loads with max. volume 127 upon start or reboot

**To Reproduce**
Steps to reproduce the behavior:
1. load pianoteq instrument. layer encoder sets volume.
2. reduce volume with layer encoder to low value (e.g. 50)
3. reboot / (or shutdown + boot)
4. pianoteq layer loads with max. volume (127, you can hear it is very loud when you play), but display at layer encoder shows previous stored volume (50). Turn volume encoder one step anticlockwise: volume display shows 49, actual volume is correct now and matches displayed value.

**Expected behavior**
Actual volume should match displayed value upon start.

**Setup:**
Im running https://os.zynthian.org/2019-09-24-zynthianos-buster-lite-1.0.0-RC1.zip which I have regularly updated.
Pianoteq: I have a license for Pianoteq stage 6.7.
zyncoder: master (c362383)
zynthian-ui: feature/stepseq (1e04e4c)
zynthian-sys: master (a5eba59)
zynthian-data: master (9cccef5)
zynthian-webconf: master (122ddd8)
in webconf: "Restore last state on startup" is active.

jofemodo, 2020-05-16:
It's a bug. I will solve it ASAP ;-)

jofemodo, 2020-05-18:
It should be solved now. Please update and test.

commit 73f4cd03 (2020-05-18): Fix #245: Fix controller value restoring (and setting) when starting Pianoteq & PD engines.
closed 2020-05-18 by jofemodo