zynthian/zynthian-ui #247: [NFR] Audio capture

le51, 2020-05-19:
Hi,
when performing, one may want to record many times in the session with zynthian built in audio recorder.
Once in the webconf "capture" section, there's a bunch of jack_capture_XX.vaw file. It can be complicated to find the famous future Top of the Pop song you have recorded during that long and crazy night ...

So, I suggest:

- to name the captured audio like "name_of_the_preset_XX.vaw"
- to show in the capture list the size of the recorded pieces of music

jofemodo, 2020-05-19:
+ mutagen should be used for wav / aiff files
+ mido could be used for mid files

jofemodo, 2020-05-19:
OK! I've added the track length to UI & webconf.
I'm thinking about the second part ...

le51, 2020-05-19:
Hi José,
thank you. Great improvement imho ! But, if it works all right on the webconf part, nothing else is displayed in the UI as you've stated.

jofemodo, 2020-09-22:
Hi @le51!

I just implement the second part of your proposal: using the active layer name for generating the recording filenames (Audio & MIDI). Please, update and test it!

closed 2020-09-22 by jofemodo

le51, 2020-09-23:
Jofemodo, my hero 😄
I can't test it now, because I'm far from my zynthian since a couple of weeks. I Will be back next month.
Cheers

commit 3b8554cd (2020-09-22): Fix #247: Fix filename counters.