zynthian/zynthian-ui #256: Collaboration guidelines needed

jypma, 2020-06-16:
In order to be most effective in helping contributors be efficient with this project, I think we can benefit from writing a few things down. A few examples come to mind:

- **Hardware scope** for supporting/testing/maintaining features: Only HifiBerry? Also USB? How many input and output channels? Raspberry pi versions (other ARM boards?), OS versions, ...
- **Software architecture**: how does `zynthian_ui` deal with threading, mutexes, logging, and performance concerns, and what has driven existing decisions?
- **Testability**: how can we start to use unit tests to help prevent regression as the number of contributors grows?
- **Software life cycle**: how do we bring features from an idea, to being tested, to being released, with potentially multiple such features developed concurrently?

By making this concrete, expectations can be managed ahead of time, both for contributors and users.

From what I've seen, some of these areas have some good ideas behind them already, but I haven't seen them written down in one place. On Github, generally a [`CONTRIBUTING.md`](https://help.github.com/en/github/building-a-strong-community/setting-guidelines-for-repository-contributors) file would capture this.

In case some of this information has already been written down somewhere else, then let's have `CONTRIBUTING.md` link there instead.

riban-bw, 2020-06-16:
Also coding styles. I know we want to avoid being too prescriptive but I feel constantly guilty about using different style to test of project even though I try to code to PEP 8.

jofemodo, 2020-06-16:
Nice ... let's do it, but i think this is a discussion best suited for the forum ;-)

jofemodo, 2020-06-16:
Please, @jypma, could you re-launch this thread on the forum, copying your starting post?

jypma, 2020-06-16:
No problem! I'll duplicate there, and we can mop together ideas from both.

[Done!](https://discourse.zynthian.org/t/collaboration-guidelines-for-github-projects/4328)

riban-bw, 2024-03-12:
Discussion moved to forum.

closed 2024-03-12 by riban-bw