zynthian/zynthian-ui #263: More consistent handling of MIDI channels for multi-channel instruments (setBfree, Aeolus)

riban-bw, 2020-07-03:
There are some synth engines that create multiple layers on multiple MIDI channels, e.g. Aelous, setBfree. These have differing behaviour to other engines and each other:

- Can only add Aelous to channels 1-4
- Can only assign first MIDI channel for setBfree. Remaining channels are chosen for you and are not necessarily intuitive
- Cannot change MIDI channel after adding layer.

**Describe the solution you'd like**
It would be advantageous for these engines to behave more like other engines, i.e. be able to add to any MIDI channel, populate consecutive MIDI channels and be able to move to other MIDI channels.

**Additional context**
I have submitted a PR (https://github.com/zynthian/zynthian-ui/pull/262) to implement population of consecutive MIDI channels for setBfree (if channels are available). There may be further improvements possible, e.g. using next available channel if consecutive channels not available (rather than current behaviour to use first available channel).
I assume Aeolus behaviour is because the engine loads to MIDI channels 1-4 by default. It is possible to change this behaviour. There is a file `~/.aeolus-presets` which includes the MIDI channel allocation, i.e. the MIDI channels listening and which _instrument_ (manual) the MIDI channel applies to . A simple solution may be to create versions of this file for each set of consecutive channels, e.g. 1-4, 2-5, 3-6, etc. (I have done this) and copy the relevant configuration before launching the engine. Another option would be to dynamically create this configuration file before launching the engine. This approach would be more flexible, allowing the engine to span different (non-consecutive) MIDI channels and maybe offering different quantity of manuals (similar to setBfree). The configuration is binary but small and fairly simple to reverse engineer. Also the source code is available at https://github.com/fugalh/aeolus so the actual file format could be deduced.

I note that moving MIDI channels is specifically disabled for these engines `self.options['midi_chan']=False`. I have not investigated what complications there are in enabling this feature but it seems plausible that implementing the features described above may facilitate adding the ability to move MIDI channels also.

jofemodo, 2020-07-03:
I totally agree with the convenience ... but it's not already done because *currently* the benefit/difficulty coeficient is too low. I mean:

+ I've partially reverse-engineered the aeolus config file. It's not difficult at all and i could consider adding MIDI-channel selection to the engine.
+ In the other hand, aeolus & setbfree need to be restarted for moving to a different MIDI channel. This is the mains reason because i didn't implement the layer's MIDI-channel change option. Of course, we could restart the engine everytime the MIDI channel is moved... although it's not a very elegant solution ;-(

riban-bw, 2024-03-12:
This is implemented in oram.

closed 2024-03-12 by riban-bw