zynthian/zynthian-ui #268: Ability to disable remote display for engines

riban-bw, 2020-07-08:
**Is your feature request related to a problem? Please describe.**
Some synth engines consume significant CPU when displaying their GUI over ssh with x-forwarding. It would be advantageous to be able to disable x-forwarding for engines whilst retaining it for Zynthian GUI.

**Describe the solution you'd like**
A configuration to disable x-forwarding and run engines headless. This could be a command line parameter / environmental variable but. It would be nice to be able to configure from webconf. (See use case at end of this request.)

**Describe alternatives you've considered**
Disable x-server . x-forwarding does not work because need to still see Zynthian GUI.

**Additional context**
If you run ssh with x-forwarding then any engines that support a GUI will attempt to display that GUI on your remote display. This will only succeed if you are running an x-server on that remote display.

This activity may increase the CPU load significantly due to the engine's GUI consuming CPU and its updates causing ssh to consume CPU.  In my experiments today using pianoteq I observed that pianoteq ran at an average of 63% CPU usage headless but 75% with the GUI displayed and ssh-server ran at 0.2% CPU headless but 10% with the GUI displayed.

Tests run with less CPU hungry engines like zynaddsubbfx give similar percentage overhead but the gross effect is obviously lower due to the scale.

As you can see, the engine itself has ~ 20% overhead which kinda makes sense (though not desirable) but ssh really ramps up using ~500% overhead and significant amounts of the poor RPi's resources.

X-forwarding was on for all of these tests with Zynthian GUI displayed on the remote display throughout.

All measurements done with same zynseq loop playing on RPi4 running Pianoteq 6.6.0 (Trial) with Steinway D Jazz preset. 2 Putty ssh connections and Xming x-server running on Windows 10 Professional.

I observe there is code to enable headless mode when there is no remote display but could not figure out how to enable this (without changing code - which is how I performed these tests). There may be an option available or easily exposed to offer this feature.

One reason I would like this is to allow more accurate testing of code development, e.g. whilst testing enhancements that may add to CPU load it would be good to see behaviour of Zynthian without this effect.
Another reason is that users (including me) may wish to use a remote display as their main display in which case they want to be able to disable remote display.

riban-bw, 2020-07-09:
Thanks @jofemodo , that works fine. Is there a chance we could add a control to web-conf for this?

jofemodo, 2020-07-10:
What do you suggest? ;-)

riban-bw, 2020-07-11:
How about a tickbox in the advanced view of User Interface below"Enable cursor". This could be a tickbox with label, "Enable remote display for engines" or similar.

riban-bw, 2020-07-12:
This seems to have stopped working for me!

jofemodo, 2020-07-22:
Strange. I didn't touch this. Could you re-confirm?

riban-bw, 2020-07-23:
I think it fails of you set ZYNTHIANX to some value but don't have X forwarding available of maybe a similar workflow. There is a message about no command set. I can't look at this today to get exact conditions.

jofemodo, 2020-09-22:
Hi @riban-bw !
Did you check this again?

riban-bw, 2020-09-23:
Errr... I think I may have been having a "senior moment". My previous comment makes little sense. This does seem to work so shall we add the configuration to webconf?

jofemodo, 2020-09-23:
If you add ZYNTHIANX envar to webconf, the variable will be defined "forever", so you would need extra changes to zynthian_engine. I would keep the ZYNTHIANX in its original way (remove the recently added "=='NONE'" condition) and would use a new envar for that:

+ ZYNTHIAN_ENGINE_X11_FWD

I will be waiting your PR ;-)

Regards,

jofemodo, 2021-08-12:
HI @riban-bw !
Should we close this?