zynthian/zynthian-ui #271: Zynthian adversely affecting jackd ability to process audio

riban-bw, 2020-07-20:
**Describe the bug**
I am able to run Pianoteq with external and internal samplerate of 48000 when Zynthian is not running without any xruns but when Zynthian is running with Pianoteq xruns occur.

**To Reproduce**
Steps to reproduce the behaviour:
1. Connect ssh client with x-forwarding to Zynthian
2. Add Pianoteq layer, C. Berchstein DG, Prelude
3. From Pianoteq GUI (shown in client's X Server) change internal rate to same as external rate, e.g. 44100, 48000, etc.
4. From Pianoteq GUI start demo playback
5. Observe regular xruns
6. Stop Zynthian
7. Launch Pianoteq and load preset C. Berchstein DG, Prelude
8. From Pianoteq GUI (shown in client's X Server) change internal rate to same as external rate, e.g. 44100, 48000, etc.
9. From Pianoteq GUI start demo playback
10. Observe no xruns

**Expected behaviour**
Zynthian should not add such load that it significantly affects engine performance.

**Setup:**
HARDWARE
Raspberry Pi 4 Model B Rev 1.2
Soundcard: Behringer UMC404HD
Display: WaveShare 7 HDMI+USB
Wiring: I2C_HWC
GPIO Expander: Not detected

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-04-09
Memory: 8% (331M/3906M)
SD Card: 37% (9.8G/28G)
Temperature: 43.0ºC

SOFTWARE
zyncoder: master (2703237)
zynthian-ui: master (8a46ae2)
zynthian-sys: master (7f30365)
zynthian-data: zynseq (548d68a)
zynthian-webconf: master (9eb269b)

riban-bw, 2020-07-21:
I will do some testing and profiling to identify which elements are consuming CPU. My top suspect is refresh and I'm not convinced this issue is new. I have always had issues running Pianoteq within Zynthian. I can run it with very low latency and high samplerate using straight ALSA. Adding JACK brings that performance down then adding Zynthian has substantial impact. I can disable new features to check if they are the cause. (Maybe a feature requests on its way to disable features from config ;-) .) If anyone has any tips on profiling then please share.

jofemodo, 2020-08-06:
Hi @riban-bw !
Have you profiled the zynthian-ui load? I think this task would be a lot more productive than #233.
Optimizing zynthian-ui bottle-necks could bring us some precious CPU cycles ...

jofemodo, 2020-08-08:
Hi @riban-bw !

I was checking this, but i couldn't see zynthian-ui going higher than 5%, and i'm not getting XRuns with any combination of engines. I disabled status bar, but it didn't make any difference ...

Could you re-check with master and see if you still see zynthian-ui consuming CPU ?

Regards,

jofemodo, 2020-10-07:
Hi @riban-bw !

I'm closing this as i can't detect any performance degradation.
Please, don't doubt to re-open if you detect some bottle-neck along your test ;-)

closed 2020-10-07 by jofemodo