zynthian/zynthian-ui #275: MIDI control mapping does not scale to LV2 paramters

riban-bw, 2020-07-22:
**Describe the bug**
Assigning MIDI CC to control Calf Monosynth filter cut-off results in no useful control because all the control is in the first fraction of the range. Calf Monosynth filter cut-off has range 10-16000Hz and I suspect Zynthian is sending values 0-127 (or maybe normalised 0-1).

**To Reproduce**
Steps to reproduce the behaviour:
1. Add synth layer: JV/Calf Monosyth > Barkline
2. Assign MIDI controller to "Cutoff"
3. Adjust MIDI controller

**Expected behaviour**
Cut-off varies from 10Hz when MIDI CC is at minimum to 16000Hz when MIDI CC is at maximum

**Actual behaviour**
Cut-off varies from 10Hz when MIDI CC is at minimum to 16000Hz when MIDI CC is just off minimum. Display of cut-off does not change.

**Setup:**
HARDWARE
Raspberry Pi 4 Model B Rev 1.2
Soundcard: Behringer UMC404HD
Display: WaveShare 7 HDMI+USB
Wiring: DUMMIES
GPIO Expander: Not detected

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-04-09
Memory: 5% (198M/3906M)
SD Card: 37% (9.8G/28G)
Temperature: 52.0ºC

SOFTWARE
zyncoder: master (2703237)
zynthian-ui: master (8a46ae2)
zynthian-sys: master (7f30365)
zynthian-data: master (8b13619)
zynthian-webconf: master (9eb269b)

**Additional context**
I considered adjusting the ttl but I think the scaling needs to be done within Zynthian so that MIDI CC 0..127 spans the lv2:minimum to lv2:maximum values defined in ttl.

jofemodo, 2020-07-22:
Sorry @riban-bw !

I misread the issue. You are right.
The zynthian controller works OK but not the assigned MIDI CC.

jofemodo, 2020-07-23:
OK! It's fixed. Please, update and test.

commit 685ab54e (2020-07-23): Fix #275: Implement logarithmic scaling on MIDI CC.
closed 2020-07-23 by jofemodo