zynthian/zynthian-ui #286: ALSA mixer shows audio outputs with index zero based

riban-bw, 2020-08-26:
**Describe the bug**
The ALSA mixer now shows all audio outputs (17 outputs for my UMC1820). These are shown as the device name plus an index, e.g. "UMC1820 Output 0". The index is zero-based but the user facing interface should really use one-based indexing (as previously discussed for other elements of the UI).

**To Reproduce**
Steps to reproduce the behaviour:
1. Configure Zynthian to use USB multichannel soundcard
2. Select ALSA mixer

**Expected behaviour**
Each audio output is labelled and indexed from 1, e.g. "UMC1820 Output 1"

**Acrual behaviour**
Each audio output is labelled and indexed from 0, e.g. "UMC1820 Output 0".

**Setup:**
HARDWARE
Raspberry Pi 4 Model B Rev 1.2
Soundcard: Behringer UMC404HD
Display: WaveShare 7 HDMI+USB
Wiring: I2C_HWC
GPIO Expander: Not detected

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-04-09
Memory: 3% (152M/3906M)
SD Card: 39% (11G/28G)
Temperature: 52.0ºC

SOFTWARE
zyncoder: master (a1cf317)
zynthian-ui: master (00dd262)
zynthian-sys: master (b62f92e)
zynthian-data: master (20e7da0)
zynthian-webconf: master (b841e35)

**Additional context**
The UMC1820 USB soundcard has (up to) 20 outputs. In the mode I use it these outputs are: 2 mono main outputs (1-2), 8 mono analogue (3-10), 1 stereo S/PDIF (11-12) and 8 ADAT (13-20). Zynthian shows these as "UMC1820 Output 0" - "UMC1820 Output 15" and "UMC1820 Output" I do not use ADAT or S/PDIF so assume the 16 indexed output channels are 3-10 + 13-20. The output labelled "UCM1820 Output" seems to adjust main outputs (1-2) which I use for my stereo mix (main monitors and headphones).

jofemodo, 2020-08-26:
It's fixed! Please, update and test ,,,

riban-bw, 2020-08-29:
This isn't fully working correctly.

1. Webconf still shows zero-based indices.
2. Deselecting all outputs in webconf still shows all outputs in Zynthian ALSA mixer panel.
3. Selecting some outputs in webconf shows those selected outputs first in Zynthian ALSA mixer panel but rest of outputs are also subsequently shown. (Probably same issues as 2.)
4. Inputs are not shown in ALSA mixer when they are selected in webconf. (Either show in mixer or do not offer in webconf.)

jofemodo, 2020-08-29:
With current implementation, showing "per-channel" controls on webconf is a problem, so i've disabled it and will show single controllers (not per-channel). When you select a multi-channel control on the webconf, you will see per-channel controls on the UI.

adf5d00

Regarding the other issues, when you don't select any mixer control on the webconf, it will show all. This is a feature ;-)

jofemodo, 2020-08-31:
And now per-channel controls are fully implemented on webconf too. Simply update and test.

closed 2020-08-31 by jofemodo

riban-bw, 2020-09-01:
Perfect - thanks!

commit dd932bb7 (2020-08-26): Fix #286: Use 1-based indexes on ALSA mixer control names.