zynthian/zynthian-ui #287: Save presets

riban-bw, 2020-08-30:
**Is your feature request related to a problem? Please describe.**
After editing synth parameters I would like to be able to save it as a new preset.

**Describe the solution you'd like**
Option to save current synth engine parameters as a new preset which will appear in the list of presets. Ability to name preset.

**Describe alternatives you've considered**
Editing an existing preset then saving as a snapshot may be similar but the preset retains its original name and it is not easily browsable.

**Additional context**
An example is to edit amsynth to create a cool lead mono synth sound. I want to save this in the preset list as "Killer mono lead". Maybe user presets could / should be stored in their own bank. Ability to edit name in webconf may be sufficient but it would be good to be able to do so in the Zynthian itself.

riban-bw, 2020-08-30:
Something is better than nothing 😁. Some non-LV2 could be done

khidr9, 2020-08-30:
That's why I'd advocate building on the existing snapshot saving, just peeling off the individual instrument parameters and saving them in a dedicated instrument level folder.  They wouldn't blend in with the rest of the presets but they could be named and called up to be re-used in other sets.  Or maybe just do that in the non-lv2 plugs and use the same interface but save the lv2 patches in native format.

jofemodo, 2020-08-31:
Mmmmm! Interesting approach ... perhaps we could have a "preset snapshot" format that we could show merged in the same view with the native presets. Users doesn't need to know about these details. Let me think about it ...  ;-)

TabulaNul, 2021-01-06:
Hi jofemodo !
anything new about saving presets ? It will be a very nice enhancement !

Thanks !

riban-bw, 2022-02-21:
Closing this issue and tracking in #614.

closed 2022-02-21 by riban-bw