zynthian/zynthian-ui #288: Ability to rename layers

riban-bw, 2020-08-30:
**Is your feature request related to a problem? Please describe.**
Each layer is displayed with its base/root engine's nickname, the engine's bank and and preset. This can result in indistinguishable layers due to similar / identical names or cryptic names which do not support user workflow, e.g. during performance. An example is creating audio layers which each use the same LV2 plugin as their base/root engine.

**Describe the solution you'd like**
It would be advantageous for each layer to have a name which could default to the current behaviour, i.e. the name of the engine, etc. This could be implemented by showing the currently derived name if the layer name is blank.
It would be advantageous to be able to change the layer name. This could be implemented using the (yet to be incorporated) onscreen keyboard - renaming panel (provided in feature/zynseq branch) and/or webconf.

**Describe alternatives you've considered**
Using a different engine for each layer but this does not provide the flexibility or clarity required and adds significant constraints.

**Additional context**
This enhancement would be beneficial for the current layer list. Some users have mentioned this in the forum, especially with regard to audio layers which may be more likely to have common base/root engine.
This would also improve the effectiveness of the (yet to be incorporated) zynmixer, allowing users to define the names for each layer that best meet their workflow, e.g. Lead vocal, Piano, Drums, etc.

le51, 2021-03-17:
+1 for that feature request

riban-bw, 2022-11-08:
This may be implemented when we refactor chain management.

closed 2022-11-08 by riban-bw