zynthian/zynthian-ui #290: MIDI routing problem with MIDI FX

MrDham, 2020-09-29:
**Describe the bug**
It looks like MIDI OUT of MIDI FX won't reach MIDI IN of synth until a workarround is applied. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a synth layer, e.g. "1#ZY >Brass / FM Thrumpet"
2. Bold click on layer "1#ZY >Brass / FM Thrumpet" and "Add MIDI FX" and choose "MIDI Event Filter"
3. Activate "pitch bend blocking" in "MIDI  event filter"
4. Play some note and move the PB wheel on channel 1 -> You hear that Pitch is bended whereas it should not be bended. 

** strange workarround **
5. Add any new MIDI Fx layer, e.g. "2#JV > MIDI Chord/Sus2"
6. Bold click on "2#JV > MIDI Chord/Sus2" and "MIDI Routing" and unselect all options. 
7. Play some note and move the PB wheel on channel 1 -> You hear that Pitch is not bended anymore. 

**Expected behavior**
In step 4,  you should hear that Pitch is not bended because pitch bend is filtrated by MIDI Event Filter. 

**Screenshots**
N/A

**Setup:**
 - SD-image
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05
Memory: 15% (143M/926M)
SD Card: 33% (9.1G/29G)
Temperature: 56.9ºC 

 - Hardware Config
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_EXTRA
GPIO Expander: MCP23017

 - Software Version
zyncoder: master (517e46a)
zynthian-ui: master (8ba25fb)
zynthian-sys: master (4ea75d9)
zynthian-data: master (20e7da0)
zynthian-webconf: master (27672ca)

**Additional context**
N/A

jofemodo, 2020-09-29:
Solved on 2552e4b.

jofemodo, 2020-09-29:
It should be solved now. Please, update and test.

commit 2a8add92 (2020-09-29): Fix #290: Fix MIDI routing, that was broke on commit d6ec6c5165937464ff7247b479d413f817f8892e.
closed 2020-09-29 by jofemodo