zynthian/zynthian-ui #293: Issue with select knob within mod-ui

gitnob, 2020-10-25:
This error is reproducable.

Creating a new mod-ui pedalboard and using "toggle switch" from MOD team for example, I can select through the control options with the select knob of my zynthian box. 
Using GX Amp X, i.e., and try to select through the list, the last line of this GX Amp X selection lets the cursor jump back to the first line. 
Similar behaviour with other plugins.

Adding a noise gate to the "gx amp x" plugin does not let me jump to the line of the noise gate plugin, because when in line 4 of "guitar amp x" the cursor is jumping back to line 1 - never reaching the configuration lines of the noise gate.

With some combinations of plugins after repeatedly toggling through the selection lines my whole GUI freezes and I cannot do anything else but rebooting the zynthian box.

When I am using a regular synth layer with multiple selection lines I can toggle through the lines without any issues.

It seems to be some problem with the mod-ui interface on the zynthian box. Web interface of mod-ui is working normally.

Any suggestion?

jofemodo, 2020-10-28:
Is this solved with the @jwoillez fix?

gitnob, 2020-10-28:
Yes, it seems to work. Thanks to @jwoillez   \o/

closed 2020-10-28 by gitnob