zynthian/zynthian-ui #297: Digital inputs from v4.1 extension kit not working

orcascc, 2020-11-11:
Hy,

the digital inputs from the extension kit are not working as expexted, see although https://discourse.zynthian.org/t/digital-inputs-from-v4-1-extension-set/4923.

I build a external box with 16 switches/keys and 4 pots to extend the control of zynthian. The switches are going to GND (tested with and without 10k pull up resistors).

After a reboot, no one of the keys is working in Zynthian.
After login to console an checking the second MCP23017 (0x21) I can see the changing values every time I press a key (1-16).
In Zynthian now the key 1-5 are working, but not 6-16.

My configuration you can see on the screenshots, for testing I produce notes with the keypress.
Zynthian is updated to the latest version (2020-11-09).

[image]
[image]

Thanks, Andy (orcas.cc)

closed 2020-11-11 by jofemodo

jofemodo, 2020-11-11:
Hi @orcascc !

I think the issue is fixed now. Please, update and test.

Although it should work OK, I recommend you clean your config file, deleting all the variables containing 3-digit (padding 0s) in its name, like this:

`export ZYNTHIAN_WIRING_CUSTOM_SWITCH_013`

or this:

`export ZYNTHIAN_WIRING_CUSTOM_SWITCH_019__UI_LONG="NONE"`

After that, reconfigure the wiring layout from the weconf.

Regards!

orcascc, 2020-11-11:
Hy jofemodo,

thanks for the update, the keys are working after I updated, clear the vars, reconfigured and checked the i2c bus (sudo watch ‘i2cget -y 1 0x21 0x12; i2cget -y 1 0x21 0x13’). Then everything works until power off, removing the power and restart. Then no one of the keys works until I check send the command above in the console again.

Maybe there is still a issue with initializing the second MCP23017 (0x21)?

regards, Andy

orcascc, 2020-11-11:
Hy,

I found another issue:
in the WebUI I changed the switches to MidiCC. I can change the Midi Number (in my example 11) and it works, but aber reload, at the switches 10-20 the Midi Number is 0.
In the /zynthian/config/zynthian_envars.sh the numbers are correct.
Below switch 10 the numbers are correct.

Thanks, Andy

[image]

jofemodo, 2020-11-11:
Solved! Thanks for reporting!

jofemodo, 2020-11-11:
Regarding the initialization problem, i just added your "trick" to the MCP23017 initialization code and it seems to work nicely after reboot. I don't understand why it's needed. The first MCP23017 works perfectly without it, and the second one (zynaptik) works randomly, or following a pattern that i don't understand. I've been reviewing the MCP23017 specs, but i can't find an explanation. 

Anyway, it seems that your trick solves the problem ... so i think we can't live without understanding everthing ;-)

Please, update and test.

Regards,

orcascc, 2020-11-12:
yes, now everything works, thants for the fast help. Great work.

best regards, Andy

commit 5ba22f08 (2020-11-11): Fix #297: Custom zynswitches doesn't work from 6 to 16. Envars have an extra leading 0 that must be removed.