zynthian/zynthian-ui #299: Channel number display of assigned CC on UI

smiths73v3, 2020-11-13:
**Is your feature request related to a problem? Please describe.**
Volume CC:7 from Channel 5, was accidentally assigned on instrument mapped to volume of instrument mapped to Channel 1

**Describe the solution you'd like**
Display of CC number on UI should display Channel:CC for example
1:7 for Channel 1 CC 7
5:7 for Channel 5, CC 7

**Describe alternatives you've considered**
none.

**Additional context**
This was caused by accidentally having the MIDI controller on the wrong channel while performing a MIDI Learn CC operation
Issue was not obvious at the time, and was difficult to determine cause of issue because MIDI channel is not visible on the UI

jofemodo, 2020-11-13:
I just implemented this, but using # as separator. The ":" is difficult to read and could cause confusion. I mean, "1:72" could be confused with "172". Also, the channel number is only shown on multi-timbral mode. It has no sense when "stage mode" is enabled.

jofemodo, 2020-11-13:
Please, update and test ;-)

closed 2020-11-13 by jofemodo

smiths73v3, 2020-11-13:
Thank you! I do not see the change though. I updated to the following and
tried all the combinations.

 SOFTWARE
zyncoder: master (465929c)
<https://github.com/zynthian/zyncoder/commit/465929c71c3cbd4200f3aae3019f226b5d633ad7>
zynthian-ui: master (11f49bd)
<https://github.com/zynthian/zynthian-ui/commit/11f49bd4990161d85c9e7608b6509966e40d9b6d>
zynthian-sys: master (0a0ca11)
<https://github.com/zynthian/zynthian-sys/commit/0a0ca11ba41373dc104d2043a166b410bfbc86d7>
zynthian-data: master (74ceb35)
<https://github.com/zynthian/zynthian-data/commit/74ceb35bfdd632a44ffbd48d68fb8d71c4f3bda3>
zynthian-webconf: master (8efc3ba)
<https://github.com/zynthian/zynthian-webconf/commit/8efc3bad3fa76899acf09ae84af46793cfdadd0e>

jofemodo, 2020-11-14:
You should be all-in-master. Remember you must disable Stage Mode (Single Active Channel) from admin menu. Displaying the channel only has sense when on Multi-Timbral mode.

smiths73v3, 2020-11-15:
Ok, I found the issue:

The two else if statements need to be swapped.
`--- a/zyngui/zynthian_gui_controller.py                                                                                                       
+++ b/zyngui/zynthian_gui_controller.py                                                                                                       
@@ -376,9 +376,6 @@ class zynthian_gui_controller:                                                                                            
                        self.plot_midi_bind("??",zynthian_gui_config.color_ml)                                                                
                elif self.zyngui.midi_learn_zctrl and self.zctrl==self.zyngui.midi_learn_zctrl:                                               
                        self.plot_midi_bind("??",zynthian_gui_config.color_hl)                                                                
-               elif self.zctrl.midi_cc and self.zctrl.midi_cc>0:                                                                             
-                       midi_cc = zyncoder.lib_zyncoder.get_midi_filter_cc_swap(self.zctrl.midi_chan, self.zctrl.midi_cc)                     
-                       self.plot_midi_bind(midi_cc)                                                                                          
                elif self.zctrl.midi_learn_cc and self.zctrl.midi_learn_cc>0:                                                                 
                        midi_cc = zyncoder.lib_zyncoder.get_midi_filter_cc_swap(self.zctrl.midi_learn_chan, self.zctrl.midi_learn_cc)         
                        if not midi_cc:                                                                                                       
@@ -386,6 +383,9 @@ class zynthian_gui_controller:                                                                                            
                        if not zynthian_gui_config.midi_single_active_channel:                                                                
                                midi_cc = "{}#{}".format(self.zctrl.midi_learn_chan+1,midi_cc)                                                
                        self.plot_midi_bind(midi_cc)                                                                                          
+               elif self.zctrl.midi_cc and self.zctrl.midi_cc>0:                                                                             
+                       midi_cc = zyncoder.lib_zyncoder.get_midi_filter_cc_swap(self.zctrl.midi_chan, self.zctrl.midi_cc)                     
+                       self.plot_midi_bind(midi_cc)                                                                                          
                else:                                                                                                                         
                        self.erase_midi_bind()                                                                                                
                                                                                                                                              
`

jofemodo, 2020-11-16:
Hi @smiths73v3 !

I just committed a set of changes that should improve quite a bit the midi-learn worflow, fixing the problems with this feature and removing some old limitations ...

Please, update and test ...

Regards,

smiths73v3, 2020-11-16:
Yes! That works great for the midi channel.
Thank you!
-S

commit bdbde79e (2020-11-16): Fix #299: Improve MIDI learning for being more consistent. Add a default mechanism for MIDI-learning to zynthian_engine base class. Modify ZynAddSubFX for using it instead of the native ML code.
Fix zynthian/zyncoder#30: Integrate changes on zyncoder for removing MIDI-learning limitations.