zynthian/zynthian-ui #305: Keyboard split

filippo-devecchi, 2020-12-07:
In zynthian is possibile to layering sound but sometime is very useful to partition the keyboard and split it. 
I think that a simple solution could be  this: 
in stagemode, leaving the feature of cloning midi to another midi channel but applying a property NOTE RANGE low and up midi note. 
So the plugin will sound only if the note is in that key range if the range is not matched no sound is trasmitted but the midi message could be route to another channel. 
In this way you can split all the key matching every situation live or studio situation jamming with looping . 
In the lower registry you can use a bass synth with arp pluing in the middle the piano with string layer (for eg.) and in the upper a lead part. 
Last request is using octave transpose in the sound. The reason is because when you split in hi range some sound could noyt be usefull so drop down or up (1 or 2) octave will be usefull for some sound.

jofemodo, 2020-12-07:
Hi @filippo-devecchi !

I just committed the implementation for this task, but as a "layer option". Simply update and bold-click a layer. You should see a "Note Range" option ...

[image]
 
Enjoy!

jofemodo, 2020-12-07:
I just improved a little bit, including the halftone transpose in the new Note Range modal:

[image]

Enjoy!

commit 75d933c6 (2020-12-07): Closes #305: Remove old transpose stuff and add it to the Note Range modal (2).
commit 458123d3 (2020-12-07): Fix #305: Change halftone by semitone on zctrl label.
commit 181f3e07 (2020-12-07): Fix #305: Improve piano plot.
closed 2020-12-07 by jofemodo