zynthian/zynthian-ui #311: Software update does not save current snapshot

riban-bw, 2021-01-12:
**Describe the bug**
If I perform a software update the Zynthian software restarts and the current layers, etc. are lost and return to the last saved snapshot.

**To Reproduce**
Steps to reproduce the behaviour:
1. Start Zynthian
2. Make some changes, e.g. add layer
3. Navigate to Admin->Software Update

**Expected behaviour**
Update completes and return to same snapshot state as before.

**Actual behaviour**
Update completes and return to last saved snapshot state.

**Setup:**
 HARDWARE
Raspberry Pi 4 Model B Rev 1.4
Soundcard: HifiBerry DAC+ ADC PRO
Display: ZynScreen 3.5 (v1)
Wiring: MCP23017_ZynScreen
GPIO Expander: MCP23017

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05
Memory: 7% (150M/1939M)
SD Card: 36% (9.3G/28G)
Temperature: 41.0ºC

SOFTWARE
zyncoder: master (aa91afe)
zynthian-ui: master (8385cd4)
zynthian-sys: master (b685fdf)
zynthian-data: master (336eadc)
zynthian-webconf: master (0587018)

**Additional context**
This may be solved by saving the current state before software update is called.

jofemodo, 2021-01-12:
I just solved as you proposed, but it doesn't work when updating from webconf or CLI. For solving this totally, CUIA should be used when restarting UI from update script.

jofemodo, 2021-01-13:
I just commited a set of changes into zynthian-sys and zynthian-ui that completely solve the problem, including updating from command line and webconf.

commit bcfca02d (2021-01-12): Fix #311: Save current snapshot before Software Update.
It doesn't work when udating from webconf or CLI. For solving this, CUIA should be used when restarting UI fromupdate script.
commit 26b3bfb3 (2021-01-13): Closes #311: Implement new CUIA "LAST_STATE_ACTION".
closed 2021-01-13 by jofemodo