zynthian/zynthian-ui #314: Native pianoteq and zynthian gui with vnc on 2 different ports

nickfielibert, 2021-01-31:
Instead of the cumbersome way to get native Ui on pc or Mac  and poor responsiveness of the same on IOS, just launch native GUI in VNC like Zynthian GUI, on a separare port. For example zynthian on ip address:5900 and pianoteq on ip address:5901

steveb, 2021-01-31:
Also the zynthian web ui could display a running pianotec in the browser with NoVNC: https://novnc.com/info.html

nickfielibert, 2021-02-01:
what do I have to do in Zynthian to get this done? Is there a novnc app for IOS to view?

riban-bw, 2021-02-01:
Displaying the GUI for an application adds overhead which increases the risk of xrun (clicks). By default engines are run headless, not displaying their GUI. It is possible to show the GUI, e.g. by logging in via SSH with X-forwarding before launching the engine. (This behaviour can be disabled.) If we always display the GUI of each engine in a virtual display we run the risk of adversely affecting the audio output.

Enabling virtual displays also has an overhead, i.e. even an idle VNC server will use some system resources.

It would be good if we could launch the GUI only when required. This would allow us to detect a remote display connection and show the appropriate GUI. Unfortunately this is not a feature of most engines, i.e. you have to start the engine headless or with GUI and cannot attach a GUI to a running instance. (Some applications do allow this, e.g. SooperLooper.)

We could add a feature to enable remote desktop access to each engine so that when it launches a corresponding remote desktop is served to which you could connect your client, e.g. VNC. This would ideally be configurable per engine (or even per instance of engine???) so that you could expose the Pianoteq UI without the overhead of running ZynAddSubFX UI. I am not sure what this would look like in the Zynthian UI. Maybe an extra step during layer creation or a configuration page in the admin menu.

This note is to describe some of the challenges and considerations of implementing such a feature. I don't answer questions here but ask some that others may wish to comment on before working on a solution.

riban-bw, 2021-02-01:
A little investigation...

- Install tigervnc-standalone-server
- Launch with `Xvnc -desktop pianoteq -once securitytypes=none :101`
- Launch synth engine with display 101: `DISPLAY=:101 pianoteq --params...`
- Connect VNC viewer to port 6001

This removes security - we may wish to retain some.

We could have a different display / port for each type of engine or for each instance of engine.

riban-bw, 2021-07-05:
Fixed! Zynthian UI is now available via VNC and the engines' GUI will show in another VNC session.

closed 2021-07-05 by riban-bw