zynthian/zynthian-ui #318: Global MIDI Reset + Destruction Action Confirmation + Undo Functionality

gr0k-net, 2021-02-18:
Bundling a few feature semi-related requests into a single post. Happy to split into several if required.

**Is your feature request related to a problem? Please describe.**

1. Global MIDI Reset - Through some fat-handedness on my own part or a bug in the software, or MIDI controller, or some combination thereof, I have occasionally had synths end up with very unusual mappings. Also, on ZynSeq I have been mapping note-on events to toggle switches, but starting a new song means having to individually unmap them one by one. A global MIDI reset would make all of this easy.
2. Destruction Action Confirmation - There are several destructive actions that do not currently have a confirmation request, and as a result I have on several occasions accidentally blown away carefully tuned effect chains etc. In my opinion all destructive actions should require confirmation.
3. Undo Functionality - related to the above, on occasion I have accidentally slipped a synth setting when moving between pages and/or instruments. Sometimes I don't even notice it has changed until later, after making further adjustments to other layers. A per-layer/channel undo function would be very handy!

**Describe the solution you'd like**

1. A global MIDI reset function, or perhaps a per-layer/channel MIDI reset.
2. A confirmation dialog when performing any potentially destructive action. This includes restarting the GUI, as it doesn't appear to save the state before restarting.
3. A global or per-layer/channel undo function, with mapable key.

**Describe alternatives you've considered**

1. As above, either global or per layer/channel
2. It may be worth having an option to turn these confirmation boxes off. Perhaps others aren't as uncoordinated as I am haha
3. As above, either global or per layer/channel

**Additional context**
I am currently on features/zynseq branch. 

I know the undo functionality is a big job, personally undo functionality is the lower priority of these three feature requests.

jofemodo, 2021-02-18:
Hi @gr0k-net !

I just added confirmation popup for the more destructive actions:

+ Clean All (recently added to main menu)
+ Remove All Layers
+ Remove Layer (from layer options menu)

I think this improves the situation. Let's see if it's needed to add more popups ;-)

jofemodo, 2021-02-18:
Hi @gr0k-net !

I just added a new layer option: Clean MIDI-Learn

It doesn't clean ZS3. Do you think it should be included too?  

Just update and enjoy!!

gr0k-net, 2021-02-20:
Amazing!!! Thank you @jofemodo! 

The confirmations have already saved me once, and the clean MIDI-Learn has also had a workout.

Thank you!

gr0k-net, 2021-02-22:
Update, I noticed last night that there is still no confirmation on remove all Audio FX. Thankfully I had intended to blow them all away, but a confirmation here too may save some heartache.

Thank you again!

jofemodo, 2021-02-22:
OK! I just added the confirmation Popup for removing All Audio/MIDI FXs.
Simply update and test.

gr0k-net, 2021-02-22:
Just updated on a fresh install on a new card, works perfectly! Thank you so much @jofemodo!

gr0k-net, 2021-02-23:
actually just discovered a little bug while playing last night. Although cancelling the confirmation dialog works, the affirmative action doesn't actually remove all effects. Apologies, should have tested more before confirming it all worked!

jofemodo, 2021-02-23:
Sorry! I can't reproduce the issue. It works perfectly for me.

gr0k-net, 2021-02-24:
hmmmm, I have just tried again.

I had 1 Noisemaker layer with 1 MIDI effect. I go to remove all MIDI effects, am asked to confirm, choose yes, but am returned to the previous screen (layer options) with the MIDI effect still in place. Thinking the screen may just not have refreshed I return to the layers screen, then back into the layer options, the MIDI effect remains.

I updated again before testing, this is a fresh stable build image on master.

zyncoder: master (ef24e0e)
zynthian-ui: master (9696069)
zynthian-sys: master (b953240)
zynthian-data: master (2028b5d)
zynthian-webconf: master (d57e6e2)

jofemodo, 2021-02-25:
Sorry @gr0k-net! My fault. It should be fixed now.

commit acf5297c (2021-02-25): Fix #318: Add missing parameter to "midifx_reset_confirmed" method.
commit d6e41812 (2021-02-25): Fix #318: Add missing parameter to "midifx_reset_confirmed" method.
closed 2021-02-25 by jofemodo