zynthian/zynthian-ui #321: ZynSeq: Add configuration for feedback tallies

riban-bw, 2021-02-22:
**Is your feature request related to a problem? Please describe.**
ZynSeq can send tallies of the play state for each sequence as a note on with state encoded as velocity. The codes are currently hardcoded for Akai APC. It would be advantageous to allow these to be configurable for different hardware controllers.

**Describe the solution you'd like**
Ability to define the velocity values for each state. Ability to define note on / note off for states. (Only if a controller uses this.)

**Describe alternatives you've considered**
Transformation using a LV2 plugin.

**Additional context**
Need to find out what codes different controllers use. If they all use the same or are configurable then this feature may not be required.

gr0k-net, 2021-02-22:
That would be incredible @jofemodo!

Being able to manually select device maps would make "dumb" controllers like my MPK Mini MK2 that can't be mapped on the fly (and only have four program modes) far more useful, as I could make a make a few different maps and quickly route midi channels via the Zynth, allowing me to control all of my instruments without having to carefully plan, pre-program, or manually swap instrument channels - which turns out to be quite time consuming when there's instrument control mappings, sequence triggers, and in my workflow, often MIDI routing layers so I can send different patterns to single instruments instances.

Having typed this out now, I could probably already do (at least some of) this with the existing MIDI mapping and MIDI profiles on the Zynth? 

Thankfully, since a friend offered me their MPD32 to use, I haven't faced this limitation, but having more precise control over mapping specific things like velocity would fix the lack of visual feedback on my controller (providing I don't use exclusion groups) and being able to quickly select profiles per device on the fly would have me using my MPK far more often.

Thank you again to both of you, not only am I loving my Zynthian, but also the community behind it. I had fun with Zynthian before even buying the hardware, and exponentially more so after I did. But since then, despite only been a few weeks (!!) it has gotten even better again, almost on a daily basis. Thank you both for all of your hard work!

gr0k-net, 2021-02-22:
Thinking about this some more and this seemed like the best place to thoughts on the subject matter, apologies if this is best left for the forum.

It was silly of me to ignore stage mode, it does make my MPK Mini much more useful than I first thought. Having per-controller profiles would still be very useful tho, as right now at least the MPD32 is my primary input. Even having the ability to assign stage / multitimbre mode per controller would allow for an incredible amount of flexibility. 

On a related note, I have just got a brand new (and allegedly much faster) SD card and installed the stable image from 2020-09-05, then updated. Starting with a fresh Zynthian image I was hoping to re-map my controller to using the free notes on channel 16 instead of channel 15 as I was using previously, thereby freeing up an entire MIDI channel. Turns out there was a reason I was using 15 instead of 16, all mappings (in WebConf / Interface / MIDI Options -> Advanced View / midi filter rules) on channel 16 are ignored.

I know that in the mapping codes 0 = 1 so 16 =15, but the Zynthian does seem to ignore mapping requests on channel 16 (or at a guess, the master channel). I have tried specifying CC and note properties, and doing global mappings, for instance:

MAP CH#15 CC => CH#15 NON

This is a verbose way of saying that per-controller assignments for multi-timbre/stage mode, being able to map by velocity, and being able to map MIDI channel 16 (CH#15 in mapping terms) might be an easy way of adding a great deal of flexibility.

closed 2024-03-12 by riban-bw