zynthian/zynthian-ui #327: ZynSeq: Show bank number on ZynPad pads

riban-bw, 2021-02-22:
**Is your feature request related to a problem? Please describe.**
It is easy to change bank without noticing. Suggest adding "Bx" prefix to each pad to indicate its bank, e.g. B1A1.

**Describe the solution you'd like**
Prefix each ZynPad sequence label with Bx where x is the bank number.

**Describe alternatives you've considered**
Have unique numbers for every sequence but that can be confusing.

riban-bw, 2021-02-22:
I disagree. I see a workflow where banks are used as songs or partials with the users switching banks during performance, sometimes needing instant switching to next / previous bank. I have been confused when the bank has changed and it would be more concerning during a performance (than during testing and development). I have implemented something that I think looks okay without adding too much to each button. I have omitted the initial "B" and used a colon to separate the bank index and the group / sequence index:

[image]

jofemodo, 2021-02-22:
OK! A "live" session is a perfect use-case for that.
But having the same number repeated on every pad and in the title bar too?? I think it's enough to have it on the title bar.
You don't get more attention from the user by repeating an element. In fact you get exactly the reverse. You create "tolerance". Your brain filter that and finally you are less aware of the redundant info.

jofemodo, 2021-02-22:
Ahhh! It's also really ugly ;-)

gr0k-net, 2021-02-23:
Adding my 2c, I also find having the bank number on the pads distracting.

The bank is already listed in the title bar and there are more important things I need to see at a glance on the pads. For me the MIDI channel and pad number are far more important. Perhaps if it was smaller and in the bottom right corner of each pad?

I do like the ability to quickly switch between banks, and agree that the two decisions are connected. The easier it is to accidentally switch between banks, the easier it needs to be to see what bank you currently have selected.

riban-bw, 2021-02-24:
I have a different opinion that the extra effort to read the label actually (subconsiously) avoids me assuming I am on a bank, i.e. I don't check the topbar for the bank and the button before I perform an action. The plan was to make the buttons uniquely identifieable (as previously suggested by @jofemodo) without adding too much faff and to provide accessible display, i.e. not just colour but also text / icons to indicate state / configuration. (People with sight impairment may not easily distinguish beteen pad colours.)

Anyway, there is resistance to this plan so I have a new one. I have developed a method of moving sequences around the bank. (Not yet exposed this feature but added a ticket to do so later.) In doing so I wanted to avoid the label changing so I added a name for each sequence that can be displayed on the pad. By default the pad gets the sequence number as its name so they appear as 1,2,3... I will add the ability to rename sequences. This may provide the best of most worlds in that you can now have any text on the pad including nothing for those preferring a sparse / bare appearance! This means that the group is not indicated so I may add a group indication in the corner to avoid the dependancy on colour or the need to manually label with group (although I may not if we think manual label is sufficient).

@gr0k-net The trigger note things is by design.

jofemodo, 2021-02-24:
Hi @riban-bw !

I understand the problem you are trying to address. I also tried to start a pad and shit!! .... why doesn't work? ... ups! I'm on bank 2! OK! The problem exist but repeating a number + semicolon on every pad is not a good solution. Have you tried to use a 8x8 pad with this? My brain is still trying yo recover from the experience ;-DD

We MUST find something better ... pleaseeeee!

riban-bw, 2021-02-26:
Okay here is what I have commited to dev branch:

Each sequence may have a name of up to 16 characters. This name is displayed on the pad, wrapped to two lines if necessary. By default the name is the pattern number assigned to the pad. You can change the name with the onscreen keyboard from the zynpad menu so can call it what you want including clearing the name completely for a sparse pad. The group is indicated by colour and a letter (A-Z) in the lower left corner of the pad which makes the bottom row of the pad consist of Group, playmode icon, play state icon.

[image]

You can make it as ugly or pretty as you like!

gr0k-net, 2021-02-28:
Sorry about the slow feedback, I haven't had much time to play this past few days.

Just updated tho, and this works great! I like to have the midi channel number obvious, so I can add that, and am using simple names like kick, bass, lead etc. I think this is the perfect compromise.

Thank you again!

closed 2021-03-12 by riban-bw