zynthian/zynthian-ui #330: ZS3 Program Change Channel Bug

gr0k-net, 2021-02-23:
**Describe the bug**
When attempting to save a patch via ZS3 program changes, the Zynthian responds to a program change on ANY channel, but only appears to SAVE the patch if it is sent on the same channel as the instrument. Although the save program screen disappears, no sub-snapshot is created.

**To Reproduce**
- Create an instrument on any MIDI channel (let's use zynthian/zynthian-sys#100)
- Select a preset and make some modifications
- Double click the Snapshot encoder, the Zynthian prompts you to send a program change message
- Send a program change on a DIFFERENT MIDI channel (let's use zynthian/zynthian-ui#122)
- The save sub snapshot screen disappears, but no snapshot is stored

**Expected behavior**
- Program change messages sent on different channels are ignored - OR
- A message is shown explaining why no snapshot was saved - OR
- ALL program change messages are accepted, irrespective of channel

**Additional context**
Currently still on feature/stepseq branch:

zyncoder: master (ef24e0e)
zynthian-ui: feature/stepseq (b47f4db)
zynthian-sys: master (6d93662)
zynthian-data: master (2028b5d)
zynthian-webconf: master (45bcb08)

jofemodo, 2021-03-03:
I suppose you are using Multi-Timbral mode. In such a case, this is the intended behaviour, more or less. You can store up to 127 subsnapshots by MIDI channel. But as you say, it would be better that PCs on different channels were ignored while learning to not cause confusion. Let me fix it ;-)

In Stage mode things works slighty different and you only have 127 subsnapshots available and MIDI channel is ignored.

Regards,

jofemodo, 2021-03-03:
Fix it! Please, update and test.

closed 2021-03-03 by jofemodo

gr0k-net, 2021-03-04:
Thank you @jofemodo!

Just updated master to and it works as expected.

Thanks again!

commit 70847096 (2021-03-03): Fix #330: When learning ZS3 on multi-timbral mode, ignore PCs not coming from the active layer MIDI channel.