zynthian/zynthian-ui #331: ZynSeq: Please add "play" in for arranger view

spurkopf, 2021-02-23:
How it says.:
Please add "play" in zynseq for arranger view

spurkopf, 2021-02-24:
Thank you for considering my views and of course they are just a suggestion.
In a short form: I think a _play_ function in the _arranger view_ would be very useful. Acoustic control about what you are doing. 
Maybe _Midi Learn_ could be sacrificed in the _arrager view_ for it? 

The _pad function_ is great to try things out, to sketch and for jamming. But in the end we always want a 3-minute pop song to earn millions with it. 
I always thought the _Arranger_ would be as intended in a classic DAW:
So several patterns simultaneously assemble linearly on a timeline and thereby arrange a complete song. Which can then be played from beginning to end. 
And I thought that it is great that _ZynSeq_ can do both. But unfortunately this is not possible in the new reincarnation. Since everything is already there for this use-case, it would be a pity if it is not used.
But as suggested - maybe I have not completely understood the overall concept of _Zynseq_.

riban-bw, 2021-02-24:
I too want this feature and have designed the system to support it. How about the SNAPSHOT encoder toggle playback of the selected sequence? If that sequence is configured for one-shot then it should play and stop promptly. I can't look at this in the current sprint which has some more urgent fixes but I will consider options and schedule implementation soon.

spurkopf, 2021-02-24:
That's great! I hope that I convinced you and not the mention of the millions ;-). _Snapshot_ would also be my suggestion. It then also corresponds to the function in _pattern mode_.

spurkopf, 2021-03-09:
Yeah, I like that, thanks for adding! 
I also like that the time display "jerks" - it looks more rhythmic ( and like these classic sequencers ).
However, I want all patterns that are in the same time position to be played at the same time.
I would also think it would be great if _play_ does not always start from the first bar instead it could begin from the selected pattern.

riban-bw, 2021-03-09:
> ...looks more rhythmic...

LOL - I don't like the update rate nor how I am driving it but have done something pragmatic to get it working.

> However, I want all patterns that are in the same time position to be played at the same time.
> I would also think it would be great if _play_ does not always start from the first bar instead it could begin from the selected pattern.

If you want to use this as an old school step sequencer then set the quantity of sequences (grid size) to 1 then add as many tracks as you want in the sequence. Set the sequence play mode to one-shot. SNAPSHOT will toggle playback for the sequence, starting or pausing at the beginning of the current bar. The screen will scroll because you have just one sequence so it is always selected. This is the exact purpose for multiple tracks in a sequence and one-shot playback mode. It provides a linear song mode for the step sequencer whilst allowing it to integrate with the looping workflows, i.e. adding other looping sequences will show their loop position whilst you play the linear song.

There is the ability to use it in another unorthodox way and have multiple _songs_, e.g. sequence 1 is your first song with as many tracks as required. Sequence 2 is your second song with as many tracks as required. You can play and edit each song independently then during performance use ZynPad to start each song as required, You can group them so that one starts when the other ends allowing cascading of songs or have them in different groups to segue between songs. I really quite like the flexibility this provides whilst maintaining a coherent set of workflows.

spurkopf, 2021-03-10:
I think the _Zynpad_ and _Arranger_ patterns were still identical in the previous version, but now I have understood that this has been changed and I have to agree with you - this opens a gate into far larger dimensions. Once again I am surprised at your ideas!
What I would still like to see would be that _play_ would be possible from a highlighted pattern - and since I have now seen that this also works with loop, it would be great if the bar range of the selected pattern could be looped. :-Unfortunately, the playhead no longer works at the moment.

riban-bw, 2021-03-10:
Playhead works here. There is now a separate playhead for each sequence so you may have one playhead looping one sequence whilst another plays through a longer sequence.

The ability to start from a specific point is a valid one but I need to think how best to do this. Maybe always start from current selection in Arranger or earlier if sequence is shorter. I would normally suggest another ticket but you might be lucky because I like the idea.

Regarding looping a region - that is not as simple, either to implement in code or to design a user workflow to identify what to loop. I can see that it _may_ look advantageous but I specifically excluded it from scope on the basis that you can create a sequence with the required pattern and loop as required and the myriad of workflows, rules and possibilities from arbitrary loop points rapidly becomes challenging to design, manage and understand.

riban-bw, 2021-03-10:
Damn! Too tempting... I have just committed to dev branch a fix for starting mid sequence. Pressing SNAPSHOT in Arranger will now toggle playback, resuming from the start of the highlighted bar or from the start of the sequence if the highlight is beyond the end of the sequence.

spurkopf, 2021-03-11:
Phew that was quick! Thanks a lot for this! We can close this feature request.

riban-bw, 2021-03-11:
Fixed in sprint 2020.02.27 scheduled to be tested and merged soon. Will close issue.

closed 2021-03-11 by riban-bw