zynthian/zynthian-ui #346: Convert onscreen keyboard class to be compliant with Zynthian schema

riban-bw, 2021-02-27:
It would be advantageous to reuse onsecreen keyboard throughout Zynthian but it is currently coded to operate within the zynseq methodology which differs from wider zynthian methodology. There needs to be a conversation about this but in the interim we should change keyboard class so that it works as a standard modal window.

riban-bw, 2021-02-27:
I will rework the keyboard class as a standard modal window and adjust zynseq code to handle this change.

jofemodo, 2021-03-01:
+1000 ;-)

riban-bw, 2021-03-06:
I've done this in dev but lost some functionality due to constraints of zyncoder library. I am looking at the library but want to avoid a dependency so may release this with the reduced functionality then fix after zyncoder refactor.

riban-bw, 2021-03-12:
Closing this ticket because I think it is done awaiting merge with testing branch.

closed 2021-03-12 by riban-bw