zynthian/zynthian-ui #350: Pianoteq Instrument Packs have changed.

bigmalletman, 2021-03-11:
**Describe the bug**
Pianoteq Instruments that I own on the Zynthian UI are being reported as demos, but they reported correctly in the actual Pianoteq 7.2.0 app.

~/.config/Modartt# grep subl Pianoteq72.prefs
VALUE name=“subl” val=“Harp:A;Steelpans:A;Celeste:A;Vibes:A;Xylo:A”

Zynthian is looking for "Steel:A" but my config has "Steelpans:A"

Also, Zynthian detects Kalimba as a dedicated Kalimba instrument pack. Kalimba is part of the Celeste package
https://www.modartt.com/celeste

**To Reproduce**
Steps to reproduce the behavior:
I am using the latest version of ZynthianOS (fresh stable version downloaded today).
I am using the latest version of Pianoteq 7.2.0. I am PRO. I own these instrument packs: Steelpans, Celeste, Vibraphones, Xylo, Harp

**Expected behavior**
Steelpans should report as "Steel Drum, Spacedrum, Hand Pan, Tank Drum". Celeste should report as "Celesta, Glockenspiel, Toy Piano and Kalimba".

**Screenshots**
https://discourse.zynthian.org/uploads/default/optimized/2X/d/d6c1c1b6076f042bdb308daa174a46053159e0c3_2_500x500.jpeg
https://discourse.zynthian.org/uploads/default/optimized/2X/5/5d41168448da5329bc12490a1d3a10cbf2378784_2_435x500.png

**Additional context**
Discourse for this issue: https://discourse.zynthian.org/t/zynthian-doesnt-realize-i-upgraded-to-pianoteq-pro-and-have-new-instruments/
Probably caused by changes from Pianoteq 6x to Pianoteq 7x.

jofemodo, 2021-03-11:
Please update and test. Perhaps you have to click the "update presets" from the webconf's Pianoteq panel and restart the zynthian UI (or reboot).

closed 2021-03-11 by jofemodo

bigmalletman, 2021-03-11:
I have tested this patch using both Pianoteq 7.0.5 and Pianoteq 7.2.0. My menu options appear correctly now.
I did not need to update presets.

jofemodo, 2021-03-11:
Nice. I hope you send some video from your Xylosynth setup. It's not something i see everyday!!! ;-)

bigmalletman, 2021-03-11:
FYI, according to the changelog for Pianoteq, 
https://www.modartt.com/pianoteq#changes
Kalimba was added to the Celeste pack on version 6.5.2 (2019/06/25); It is apparently a one octave diatonic instrument.

jofemodo, 2021-03-11:
Then the instrument pack fir the Kalimba has been bad labeled from the beginning. Ups!!
We needed someone with this package licensed to find the bug ;-)
It's fixed now for all versions.

commit b9860831 (2021-03-11): Fix #350: Rename some instrument packs for v7.
commit df6f2b7a (2021-03-11): Fix #350: Fix kalimba's instrument pack name for v6.5.2 and later.