zynthian/zynthian-ui #351: Pianoteq 7.2.0 PRO appears as STANDARD in the Zynthian UI

bigmalletman, 2021-03-11:
**Describe the bug**
I am a PRO user. I appear as a PRO user in the Pianoteq UI, but Zynthian is reporting me as STANDARD. There does not appear to be a separate binary for PRO, or PRO reports itself differently now.

 /zynthian/zynthian-sw/pianoteq6/arm-32bit/Pianoteq\ 7\ PRO
 does not exist, it is just
 /zynthian/zynthian-sw/pianoteq6/arm-32bit/Pianoteq\ 7

root@zetes:/zynthian/zynthian-sw/pianoteq6# ./pianoteq --version
Pianoteq version 7.2.0/20210212 – http://www.pianoteq.com/pianoteq
Copyright © 2021 Modartt.

in that Pianoteq72.prefs file…
VALUE name=“LKey” val="# Pianoteq key file (v7) # Product: Pianoteq 7 PRO # Licensed to: Daniel Wells # Computer: zetes # Date: Wed Mar 10 21:36:07 2021 …

root@zetes:/zynthian/zynthian-sw/pianoteq6# echo $PIANOTEQ_PRODUCT
STANDARD
root@zetes:/zynthian/zynthian-sw/pianoteq6# echo $PIANOTEQ_VERSION
7.2.0
root@zetes:/zynthian/zynthian-sw/pianoteq6# echo $PIANOTEQ_TRIAL
0

**To Reproduce**
I own Pianoteq PRO 7.2.0. I just downloaded it from https://www.modartt.com/. It is possible how Pianoteq reports STANDARD vs PRO has changed. There is not separate binary as far as I can tell.

**Expected behavior**
Although I don't think I'm missing any features, Zynthian should appear as PRO, not STANDARD

**Screenshots**
https://discourse.zynthian.org/uploads/default/optimized/2X/5/5d41168448da5329bc12490a1d3a10cbf2378784_2_435x500.png
https://discourse.zynthian.org/uploads/default/optimized/2X/2/2621bd73c6b9ae3825f73933fc731c895f826a61_2_690x150.png
https://discourse.zynthian.org/uploads/default/original/2X/a/a3305060b37ec97e421001a3abf29a1024c337e5.png

**Additional context**
Discourse for this issue: https://discourse.zynthian.org/t/zynthian-doesnt-realize-i-upgraded-to-pianoteq-pro-and-have-new-instruments/

jofemodo, 2021-07-05:
It seems like Moddart changed the ID string format when running " ./pianoteq --version". Or perhaps they are using the same binary file for STANDARD and PRO. For solving this we should look for the version on the prefs file instead of using "pianoteq --version".

mheidt, 2021-10-29:
Ah damn, @jofemodo 
the LKEY only exists after activation...chicken/egg problem.

mheidt, 2021-10-29:
@dcoredump 
Do you remember, why you coded the product (PRO, STAGE) into the pref file name?

dcoredump, 2021-10-30:
Unfortunately not. That was so long ago now that I can't remember. I havn't touchec the code since then either. But actually it should not be so problematic to change this.

mheidt, 2021-10-30:
Ok. 
@bigmalletman 
Could you test the code with just the line
PIANOTEQ_CONFIG_FILENAME = "{}.prefs".format(PIANOTEQ_NAME)
No "if" against version 6.
Do you want to do a PR afterwards or should I do it?

mheidt, 2021-10-31:
@bigmalletman 
It's not going to be that easy. I tested it with my binary (STAGE). The issue seems to be, that pianoteq is creating the prefs file before the activation with a STAGE in the name.
So...when you deleted all 74 prefs and uploaded a new binary. What is the name of the prefs in your case? without PRO? Wonder how it looked like with a STANDARD version.

bigmalletman, 2021-10-31:
I don't entirely understand the question. The binary is the same between Standard and Pro. Your activation determines which features you get, not the binary. My webUI is currently showing STANDARD, even though I have PRO activated, and I'm not sure how to get it to show PRO again like before.
To me, it seems reasonable for the workflow to check the version each time the sw-pianoteq page is visited.
Alternatively, remove the "Product" information altogether. Since Zynthian does not directly use any of the additional features of higher products, it might be reasonable to only present to the user that Pianoteq is activated.

mheidt, 2021-11-01:
What I learned the last time is, that it's not on us to decide how to name the .prefs file. During the first run Pianoteq itself creates the file. We just have to adapt.
Holger knew during his first implementation, that for STANDARD, the prefs file was generated without the product info (STAGE or PRO) and for STAGE it was shown.  We could only test with STAGE and the Trial version (STANDARD), because we didn't know any person having a PRO license. That's why he assumed (and I would have as well), that PRO should show as well.
Now we know, that PRO is not showing. That's why I changed the code like in my last statement.
Could you test that code?

bigmalletman, 2021-11-01:
I feel like I must not be communicating well. Here is the sentence I would like to highlight:
<pre>My webUI is currently showing STANDARD, even though I have PRO activated, and I'm not sure how to get it to show PRO again like before.</pre>
[image]
Here are the relevant code segments:
[image]

[image]

After I uploaded my binary, the sw-pianoteq page displayed at "STANDARD". When I activated the product, the page did not change from "STANDARD" to "PRO", even though my LKey contains "pro".

<pre>VALUE name="LKey" val="# Pianoteq key file (v7)&#10;# Product: Pianoteq 7 PRO&#10;# Licensed to: Daniel Wells&#10;# Computer: zetes&#10;# Date: Fri Oct 29 17:08:06 2021&#10;&#10;</pre>
Please note that the val contains control code "10" for "Line feed", but github is displaying the above as new lines.

I cannot seem to get the product displayed in the webUI to change from "STANDARD" to "PRO".
So to test, I added this line...
[image]

However, the webUI still does not change from "STANDARD" to "CHEESE".
When does the get_pianoteq_binary_info() function actually get called?? Since the product type is determined by the activation, not the upload, it needs to be called more often.

mheidt, 2021-11-01:
OK,
Please try adding those two lines in the webconf:
[image]

```
		else:
			self.pianoteq_autoconfig()
```

bigmalletman, 2021-11-04:
Okay... everything _seems_ to be working here.
[image]
[image]
[image]

mheidt, 2021-11-04:
Great.
Once the PR is merged into testing, you could test it a final time before we close this issue.
Thanks,
Markus

bigmalletman, 2022-04-09:
What is the status of this one? Apparently my zynthian has had a lot of updates since last time I used. Now Pianoteq75 PRO.prefs is back. Everything is appearing as demo instruments. Pianoteq isn't even launching. I was hoping to use my zynthian at a rehearsal in a couple hours, but that's not going to happen. I need to figure out how to turn off updates, because my entire setup is dead now.

jofemodo, 2022-04-15:
Not yet. We hope to release a new stable very soon, including all the latest changes.
For the future, we will change the way big changes are developed and tested. We have learned a few lessons in the current development cycle.

riban-bw, 2022-11-07:
Stable release fixed this.

closed 2022-11-07 by riban-bw