zynthian/zynthian-ui #357: Include key-range (transposing/split) info into the ZS3s

bigmalletman, 2021-03-16:
**Is your feature request related to a problem? Please describe.**
Currently, layer transposition is not part of subsnapshot. In order to have multiple instruments with different keys, you must either change instruments and change the layers transposition manually or you need to add midi-fx transposition to each subsnapshot. (There may be other ways as well).

**Describe the solution you'd like**
Subsnapshots would include the layer transposition/split details.

**Describe alternatives you've considered**
Currently, I add midi-fx transposition to each subsnapshot.

**Additional context**
Discourse for this feature request: https://discourse.zynthian.org/t/per-instrument-transposition/5560
I have a work around that seems to work for my needs, but it was suggested that I create a feature request.

jofemodo, 2021-03-16:
I's done. Please, update and test.

closed 2021-03-16 by jofemodo

bigmalletman, 2021-03-16:
I discovered this issue in testing this feature.
https://github.com/zynthian/zynthian-issue-tracking/issues/396

bigmalletman, 2021-03-16:
I also discovered this more-urgent issue in testing this feature.
https://github.com/zynthian/zynthian-issue-tracking/issues/397