zynthian/zynthian-ui #358: Volume and dynamic etc options change after transposing layer

bigmalletman, 2021-03-16:
**Describe the bug**
Possible bug discovered in testing of new feature enhancement implemented in https://github.com/zynthian/zynthian-ui/issues/357
The options defined by the top left and top right knobs change after transposing a layer.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a new layer for Pianoteq and select an instrument.
2. Observe the options for volume ("95") and dynamic ("64").
3. Go back and back and to go layer options to transpose the layer down an octave
4. Return to layer
5. Volume and dynamic values have changed. In my case, to "48" and "21".

This also occurred when I did the same for a SF2 instrument. In that case, values for volume and pan were changed.

**Expected behavior**
Options should not be affected by changes in key/range.

**Screenshots**
Step 2: https://wellsd.com/misc/zynth/zyth1.jpg
Step 3a: https://wellsd.com/misc/zynth/zyth2.jpg
Step 3b: https://wellsd.com/misc/zynth/zyth3.jpg
Step 5: https://wellsd.com/misc/zynth/zyth4.jpg

**Setup:**
Raspberry Pi 4 Model B Rev 1.2
Soundcard: HifiBerry DAC+ ADC PRO
Display: ZynScreen 3.5 (v1)
Wiring: MCP23017_ZynScreen
GPIO Expander: MCP23017
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05

**Additional context**
I am testing feature implemented and discovered this bug.
Discourse for this issue: https://discourse.zynthian.org/t/per-instrument-transposition-solved/5560/
Feature request: https://github.com/zynthian/zynthian-ui/issues/357

jofemodo, 2021-03-16:
Ups! You found a nice bug. It's solved!
Please, update and test.

Thanks!

closed 2021-03-16 by jofemodo

bigmalletman, 2021-03-16:
Added Pianoteq Glockenspiel layer.
Changed to transpose 2 octaves.
PASS: Values for volume and dynamics did not change.

Added SF2 Crotales layer.
Changed to transpose 2 octaves.
PASS: Values for volume and pan did not change.

Thanks.

commit 1ebf2614 (2021-03-16): Fix #358: Avoid instrument controllers to be changed when accessing note-range modal.