zynthian/zynthian-ui #360: Pianoteq LV2 fails if it can't show its UI

riban-bw, 2021-03-18:
**Describe the bug**
Launching Pianoteq LV2 will produce no sound unless there is a graphical environment to display its output, e.g. VNC.

**To Reproduce**
Steps to reproduce the behavior:
1. Disconnect ssh with x-forwarding
2. Disable VNC
3. Create layer with Pianoteq7 Stage LV2

**Expected behaviour**
Playing will produce audio.

**Expected behaviour**
Playing produces no audio.

**Setup:**
HARDWARE
Raspberry Pi 4 Model B Rev 1.4
Soundcard: HifiBerry DAC+ ADC PRO
Display: ZynScreen 3.5 (v1)
Wiring: MCP23017_ZynScreen
GPIO Expander: MCP23017

SYSTEM
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05
Memory: 43% (849M/1939M)
SD Card: 39% (11G/28G)
Temperature: 47.0ºC

SOFTWARE
zyncoder: master (ef24e0e)
zynthian-ui: master (7084709)
zynthian-sys: master (cdbb0e9)
zynthian-data: master (10db183)
zynthian-webconf: master (25b1ace)

**Additional context**
LV2 works if VNC or x-forwarding is enabled. Log shows Zynthian attempting to launch jalv.gtk when it should be launching jalv.

jofemodo, 2021-03-19:
I can't reproduce this issue. I tried with different sequences of steps, including yours, but it works for me.
BTW, you can now enable/disable the VNC server and layers will be restarted with/without GUI. It's implemented by saving/restoring the "last state snapshot" ;-)

riban-bw, 2021-03-19:
Since the change to restart engines when toggling VNC behaviour has changed. Engines UI no longer open on SSH X-forward session which is unexpected but probably advantageous.

I can't now reproduce this so probably fixed by that change.

closed 2021-03-19 by riban-bw