zynthian/zynthian-ui #374: "Layer option" workflow problem when pushing "Back" and "Select" buttons

MrDham, 2021-04-06:
**Describe the bug**
When editing a "Layer Options", for example "Note Range & Tranpose", pushing "Back"  and "Select" leads you to 1st layer's "Control" screen intead of  "Note Range & Tranpose" again. It works the same with all options ("Midi Routing", "audio routing", ... )

**To Reproduce**
Steps to reproduce the behavior:
1.  Create a layer
2. **Bold Select** layer 1 -> You are in "Layer 1 Options"
3. **Select** "Note Range & Tranpose" -> You are in "Note Range & Tranpose"
4. **Back** -> You are back to "Layer 1 Options"
5. **Select** "Note Range & Tranpose" again -> you are in1st layer's "Control" screen 

**Expected behavior**
5. **Select** "Note Range & Tranpose" again -> You are in "Note Range & Tranpose" again

**Screenshots**
I don't have. I Can try to get them if necessary.

**Setup:**
 - SD-image
 Build Date: 2020-09-05 

 - Hardware Config
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_EXTRA
GPIO Expander: MCP23017

 - Software Version
zyncoder: stable (d852c8f)
zynthian-ui: stable (01c808c)
zynthian-sys: stable (df05b71)
zynthian-data: stable (194fd59)
zynthian-webconf: stable (dec5f31)

**Additional context**
N/A

jofemodo, 2021-04-06:
I just pushed a patch that solve a bunch of problems related with back action. Please, update and test (testing branch).
This commit should be tested carefully and passed to stable (hotfix)

closed 2021-04-06 by jofemodo

MrDham, 2021-04-07:
Quick test -> It is fixed :+1:  

`This commit should be tested carefully and passed to stable (hotfix)`
OK, I'll stay in testing branch for a while.

commit 1894310c (2021-04-06): Fix #374: Fix several problems related with back action.