zynthian/zynthian-ui #377: Show range for MIDI learn

riban-bw, 2021-04-17:
**Is your feature request related to a problem? Please describe.**
Some MIDI learned controllers show ERR when MIDI control values are sent. I assume this is because out-of-bound values are being sent. There is no (obvious) way to see what the range of permissible values is.

**Describe the solution you'd like**
During MIDI learn display the range (min-max) of permissible values. Indicate (maybe with a dot or other symbol) if the range is enumerated rather than continuous.

**Describe alternatives you've considered**
Document the range of permissible values.

**Additional context**
I have not proven that the ERR indication is the result of out-of-bound values, this is an assumption. An example controller that exhibits this issue is ALO (looper) Enable control. I have tried sending 0/127 and 0/1, any of these values trigger the ERR indication.

riban-bw, 2021-04-17:
I gave an example in the original report: ALO's Enable control.

jofemodo, 2021-05-03:
Ups! Sorry, i didn't fully read the additional context.

riban-bw, 2022-11-07:
This seems to be fixed - probably by some reworking of the zcontroller.