zynthian/zynthian-ui #379: UI blocked after removing one audio effect in a layer

MrDham, 2021-04-26:
**Describe the bug**
UI is blocked in "layer option" menu after removing an audio effect in a layer

**To Reproduce**
Steps to reproduce the behavior:
1. Create a layer e.g. 1#ZY > Arpeggios/Arpeggio1
2. **Bold Select** layer 1 -> you are in "1#ZY Options"
3. **Select** "Add Audio-FX" and choose any effect, e.g. "ZynChorus"
4. **Back** until "layers" menu apears
5. **Bold Select** layer 1 -> you are in "1#ZY Options"
6. **Select** effect added in step 3
7. **Select** "Remove Audio-FX"  -> You are blocked in "1#ZY Options"

Only way out : reboot with an external device (e.g. webconf on an external PC)

**Expected behavior**
Normal navigation in UI after step 7

**Screenshots**
Can provide if needed

**Setup:**
 - SD-image
Raspbian GNU/Linux 10 (buster)
Build Date: 2020-09-05 

 - Hardware Config
Raspberry Pi 3 Model B Plus Rev 1.3
Soundcard: HifiBerry DAC+
Display: PiScreen 3.5 (v2)
Wiring: MCP23017_EXTRA
GPIO Expander: MCP23017

 - Software Version
zyncoder: staging-2104 (d852c8f)
zynthian-ui: staging-2104 (72a5241)
zynthian-sys: staging-2104 (d90d6f1)
zynthian-data: staging-2104 (f71259d)
zynthian-webconf: staging-2104 (153a020)

**Additional context**
Testing "staging-2104"

jofemodo, 2021-04-26:
Hi @MrDham !

It should be solved now. Please, update and test.

Thanks a lot!

closed 2021-04-26 by jofemodo

MrDham, 2021-04-27:
I just tested, it seem to be ok. Thanks.

commit b2f6a7cf (2021-04-26): Fix #379: UI blocked after removing one audio/MIDI effect in a layer.
commit b3a139c5 (2021-04-27): Fix #379: UI blocked after removing one audio/MIDI effect in a layer.