zynthian/zynthian-ui #381: Adding a virtual "favourites" bank folder for synths

Pastitas, 2021-05-11:
**Is your feature request related to a problem? Please describe.**
The problem is the favorites list (currently bound to bold press snap) seems to be easily forgotten and, because of that, It's less often used.

**Describe the solution you'd like**
I'd like a bank folder at the top of the preset list called "Favorites", maybe even separated by a "------" from the rest, where all the favorites from a synth appear. This would maintain the current list access via the bold press, just add another list for enhancing usability. 

This might even allow for another "default" folder like the favorites one with some community voted presets to show what the synth is capable of.

**Describe alternatives you've considered**
Another alternative would be adding some kind of text that reminds the user of the hidden favorites menu, but this seems unpractical and more complex than just presenting said list as a folder

*

Pastitas, 2021-05-12:
So quick, It works beautifully!
I'm trying to understand the changes in the code, you just added a new bank which comes from the favorites list, right?

jofemodo, 2021-05-12:
In fact code is simpler like this. Having a virtual "favorites" bank is more straight forward than having the "view only favorites" mode, that was implemented before.

closed 2021-05-12 by jofemodo