zynthian/zynthian-ui #383: Improve UI layout

jofemodo, 2021-05-21:
**Is your feature request related to a problem? Please describe.**
UI layout can be improved with some simple changes.

**Describe the solution you'd like**
We would proceed step by step:

1.) Remove selector widget in most (or all!) menus.
2.) Study the possibility of reducing the size of right column on selectors
3.) Study alterntives to the current "activity" animated icon: smaller, change position, etc.
4.) Study how to expand the status area, so we can display more info, or use bigger fonts, bigger VU-bars, etc.
5.) etc.

jofemodo, 2022-08-26:
3 & 4 remain undone:

3.) Study alterntives to the current "activity" animated icon: smaller, change position, etc.
4.) Study how to expand the status area, so we can display more info, or use bigger fonts, bigger VU-bars, etc.

riban-bw, 2022-11-07:
Closing this ticket as issues should be tracked in main issue tracker.

closed 2022-11-07 by riban-bw