zynthian/zynthian-ui #386: Allow Zynthian to be operated blind

riban-bw, 2021-05-30:
**Is your feature request related to a problem? Please describe.**
There are many musicians with impaired vision or are blind who may struggle or be unable to operate Zynthian.

**Describe the solution you'd like**
It would be advantageous to provide an audio description to assist with navigation and adjustment of the device. This should output audio via a secondary audio pretty, e.g. headphone. During navigation, each screen title could be announced. During list navigation the current selection could be announced. During parameter adjustment the current value cold be announced. Selecting a parameter adjustment panel could describe the parameters and values for each encoder.

**Describe alternatives you've considered**
A screen reader with mouse control could read screen elements.

**Additional context**
There may be elements that do not read accurately sure to inappropriate code names or poor interpretation. It may be advantageous to have optional fields for screen elements specifically for accessibility phrases. The Linux command espeak can provide text to speech and is available from the Debian repository. I have performed a proof of concept hack using espeak which was okay. It may prove advantageous to write our own wrapper for the espeak library. I have used this for similar solution to enhance access to other products.

We may wish to offer configuration like choice of voice, speed of replay, etc.

jofemodo, 2021-05-30:
I think festival (or flite) gives a better voice synthesis and they are available on the debian repo too.
Let me give them a quick try. Implementing some voice assistance on screens and menus doesn't looks as a difficult task. Of course, devil is on details ;-)

riban-bw, 2021-05-30:
I just gave flite a spin and the voice is more natural but I found it a bit less clear. It may be possible to make it more inteligable by configuration, e.g. it was low bandwidth so sounded muffled and had an American accent. I suggest creating a wrapper for each library to give a uniform interface and integrating with Zynthian. We can then chose whichever best suits the user.

riban-bw, 2021-06-01:
I think I tried festival when I was building a module to provide access for blind and partially sighted users to a control system but it wasn't suitable and we used espeak which was well recieved for its intelligibility. Maybe flite wasn't available at the time. Flite and espeak install easily on the Zynthian and both seem to run fine. I understand thta @jofemodo has progressed the PoC further so I will wait to see his effort rather than duplicate effort.

I suggest some features:

- Abilitiy to enable / disable easily - maybe with a dedicated (USB QWERTY) keyboard shortcut to toggle and / or multiple switch presses on main unit (like a three finger salute)
- Ability to enable at boot - maybe holding a switch during startup
- Announce title of screen when opened
- Announce list entry when highlighted
- Anounce value of encoder when adjusted
- Anounce value of control when screen touched - this needs some consideration because it may interfer with current behaviour
- Announce value of control when encoder touched (not pressed) - this requires significant hardware redesign and may prove technically challenging and impracticable
- Configuration for voice, speed, verbosity, etc.
- Read existing UI text except where this is cryptic or reads wrong where there could be a secondary field to hold announcement text or maybe the known cryptic elements are automatically converted to readable text algorithmically
- Design principles written to ensure future development will be accessible
- Revisit documentation to ensure it is accessible
- Report inaccessible elements as separate github tickets (after initial development has reached stable release)

riban-bw, 2021-06-04:
I can't get this to work. I have switched to "sight impared" repo on ui and sys and performed update.

riban-bw, 2021-06-04:
There are some features needing implementing:
- Read out confirmation menu
- Allow speed of playback to be configured
- Not obvious menu title is being read, maybe append "menu" or "screen" to spoken title and add pause between title and menu item
- Not reading capitalised initials, e.g. MDA Piano
- Obxd is not capitalised so above would not resolve this
- Would be advantageous to be able to repeat announcement. Could be added as a function that can be assigned to a button, e.g. S4
- No indication of what is on each encoder in control pages
- No indication of value as adjusting encoder
- "Clone MIDI to" does not pronounce "to" correctly
- It is not obvious that you are in a selection list
- No anouncement in Note Range & Transpose screen
- Announcing "&" as apersand
- Send Audio to... spells out names of outputs (maybe we can use alias here which will improve UI for all. This is a common issue though. It looks like it spells a word if it includes puntuation or numbers, e.g. "AubioNotes(Audio2MIDI)"
- There can be long delay between operations which are silent. Maybe add periodic beep to indicate progress analogous to the annimated icon
- Title of each pages of controls can be cryptic, e.g. Manual I (0). Could prefix with "control page" or similar
- Many elements are cryptic, e.g. the title of each page of controls - may need to provide improved training / manual
- Sequencer not accessible (I can guess your response to that!!!)
- "Start recording" does not pronounce correctly
- "Isn't checked" is sometimes not clear, especially without context. Maybe "Not checked" would be better.
- "Restart UI" does not pronounce correctly
- Content of network info page is not announced
- "QmidiNet" not pronounced correctly
- "Preset Preload" not pronounced correctly (both words sound wrong)
- "=> Multi-timbral Mode" announces "equal greater than..."
- "RBPi Headphones" not pronounced correctly

jofemodo, 2021-06-06:
Jojojo! You wrote only a fraction of all needed fixes, but it's OK for a proof of concept ;-DD
But yes ... lots of improvements to do and "special cases" we need to take into account.

And yes, i would be nice to have an alternate engine for espeak. Festival is far from perfect.

Anyway, i think i could reduce this list by a half in a few days.

Thanks!

riban-bw, 2021-06-08:
I hit a couple of hitches with my implementation of espeak so will pause that for a while.

I did add an admin menu option to select voice assist which toggled between None | Festival | Espeak. It set the value of `sight_impaired_enabled` to the name of the speech engine, e.g. "festival" or None if not used. This means very little change to code already implemented which checks for sight_impaired_enabled as a Boolean so will differentiate between None and something else. This was done on a unit that is not accessible to me now and may have included some other, incompatible changes so I suggest maybe leaving this to @jofemodo to implement. (Unless he wants me to add it to code and push to git.)

The list I made was based on my experience of trying to use it. There are a few items in the list that need to be resolved for the PoC, e.g. confirmation screens must be read out. Without this navigation is almost impossible as you are left with no indication of the current state or navigational location.

There were some functions that I would have liked more context but I realised that it is the same thing for a sighted users, i.e. the user needs to understand how the device works to some extent.

I encountered xruns whilst testing this. I wonder whether this is caused by use of the rather inefficient Raspberry Pi onboard sound or maybe alsa is struggling with two different audio clocks.

riban-bw, 2022-11-07:
Now tracking in main issue tracker: https://github.com/zynthian/zynthian-issue-tracking/issues/820

closed 2022-11-07 by riban-bw