zynthian/zynthian-ui #388: Capture Playback via Effects Chain

gr0k-net, 2021-06-29:
**Is your feature request related to a problem? Please describe.**
I would love to be able to play captures and field recordings uploaded to the Zynthian through an effects chain.

**Describe the solution you'd like**
Ideally you could long press on a capture, and choose the output in a similar way you can do with existing layers. In this way you could choose system out left / right, an effect layer, or some combination of the two.

This long press options screen would allow the possible addition of other capture related features, for example assigning MIDI Notes or CC messages to trigger playback (another feature request I will add separately).

**Describe alternatives you've considered**
Depending on how everything is hooked up behind the scenes, another possible option could be a (new, or modified existing) sample playback plugin/instrument/effect that allows the selection and MIDI synced playback of files from within the Zynthian UI.

**Additional context**
My specific context is wanting to manipulate field recordings uploaded to the Zynthian, but there are other instances where this could be useful, for example overdubbing on the Zynthian, filtering of existing captures, adding pre-recorded textures to existing tracks without requiring another device, etc.

jofemodo, 2021-06-29:
It's implemented on development branch "feature/388_audio_player_routing".
Please, test and report.

gr0k-net, 2021-06-30:
Incredible, thank you @jofemodo!

I've only had a quick play around, but so far it seems to work perfectly! 

At first I was trying to long press on the capture itself rather than in the main menu, but it does makes more sense where you put it as things will quickly get confusing if different captures are coming out of random places all the time.

(I can see the possibility of being able to play back multiple captures at the same time possibly having creative uses, but probably an edge case, and not required for what I'm doing atm)

I'll be filtering and re-sampling/re-recording, and I have already tried this on the Zynthian using your update, and it worked perfectly! This means overdubbing is now a possibility on the Zynthian, so cool!

Thank you again!!!

gr0k-net, 2021-06-30:
I have just noticed one small bug:

When switching back to the audio recorder screen whilst a capture is playing the audio recorder volume control does not reappear (only the Zynthian logo is visible) and if the previous page was an instrument panel (I use the buttons on the front to quickly swap between instrument / sequencer / mixer / audio recorder) the encoders continue to control the parameters on the last instrument page you were on.

I'm not sure if this small bug is specific to this branch, or if this issue existed in stable too, as I have never swapped between the two screens so frequently before.

gr0k-net, 2021-06-30:
Also, happy to create another feature request if required, but on the Audio Recorder screen, on the right hand side below the volume control or logo is a spare spot for another control.

It would be super useful to an _hh:mm:ss_ ticker there during playback and recording.

It would be even cooler if we could use the encoder next to it to skip forwards and backwards during playback (rather than scrolling between captures). From the Audio Recorder screen could work like this:

- turn select encoder to scroll to capture
- press select encoder to play
- select encoder scroll is locked
- - turning it right advances the playhead 1% per click (or other appropriate value, perhaps working with speed/velocity of encoders if possible?)
- - turning it left to rewind the playhead 1% per click (or other appropriate value, perhaps working with speed/velocity of encoders if possible?
- press select encoder again to stop play
- select encoder can now be used to scroll again

Doing this would mean you have to stop one capture before starting another, but that is a minor issue as only one capture can be played at a time anyway. The only other downside would mean you have to stop playing a capture to toggle loop status, tho in most cases I imagine the decision to loop a capture would be decided ahead of time anyway.

Apologies, I don't want to be greedy! Just passing on some thoughts as I play with the new feature, which I'm already having a heap of fun with.

Thank you so much!

riban-bw, 2021-06-30:
I would prefer to keep ability to start a recording without stopping the existing and adjust loop mode. Maybe use another encoder. Anyway, defo out this as a separate request.

gr0k-net, 2021-06-30:
Aaaaah, very good point @riban-bw, I hadn't realised locking from scrolling would also prevent starting a new recording, and hence overdubbing!

I will create a new feature request for this feature

jofemodo, 2021-06-30:
The volume control "not -showing" when re-entering the audio recorder is an old bug that i discovered yesterday while implementing this task. I will solve it ASAP, same that the playhead time, as soon as i find a good solution for the rewind/advance workflow,

Regards,

closed 2021-07-01 by jofemodo

jofemodo, 2021-07-01:
OK! The volume control issue is fixed.

gr0k-net, 2021-07-01:
Thank you @jofemodo,

Have just moved to testing, and the new features continue to work, and the volume control is back.

Thanks again!!

commit bf3b5b61 (2021-07-01): Fix #388: Show volume control when re-entering audio recorder.