zynthian/zynthian-ui #412: XRUNs caused by vcgencmd

davidfurey, 2021-11-27:
I've been investigating if the Raspberry Pi 4B (4GB) running Zynthian can manage a Jack period of 128 frames without xruns. My focus has been on the Salamander Grand Piano using LinuxSampler.

I'm using the InnoMaker HiFi DAC Audio Hat with jackd options `-P 70 -t 2000 -s -d alsa -d hw:1 -r 44100 -p 128 -n 2 -X raw` and RBPi Audio disabled.  With this configuration I experienced regular xruns.

I used [trace-cmd](https://www.trace-cmd.org/) on the Pi and [Kernel Shark](https://kernelshark.org/) offline to explore the cause of the xrun.  In the screenshot below you can see the regular pattern of real-time priority processes running each time Jack processes a period.  This pattern is interrupted around the cursor position. The interruption is caused by a fork system call, seen as the highlighted `copy_process` event. 

As far as I can tell, `fork` blocks all threads, even those with real-time priority.

[image]

Using [GDB](https://www.gnu.org/software/gdb/) I was able to identify the culprit as the fork system call required by the regular poll of `vcgencmd` to check for undervoltage and overtemp events. Once I commented out this section from `zynthian_gui.py` the xruns stopped:

```
try:
        # Get ARM flags
        res = check_output(("vcgencmd", "get_throttled")).decode('utf-8','ignore')
        thr = int(res[12:],16)
        if thr & 0x1:
                self.status_info['undervoltage'] = True
        elif thr & (0x4 | 0x2):
                self.status_info['overtemp'] = True

except Exception as e:
        logging.error(e)
```

In fact, I can now run the Salamander grand with a period of 64 frames with almost no xruns.

I'm sure there are a few ways we could retain this feature without triggering a fork.  One option would be to run an entirely separate process that regularly runs vcgencmd and puts the output in a file to be read by zynthian-ui.  

I do wonder though, if a better approach might be to pull the real-time priority jack processes out of zynthian-ui instead. This would make the performance of the zynthian-ui app less critical. 

Since I haven't spent a lot of time studying the zynthian software architecture I would appreciate others input before spending any time on even the simpler solution I've suggested.

jofemodo, 2021-11-27:
Ohhh!! This is a very interesting research. Let me test your fix, but it looks promising!! Congrats and thanks!!

jofemodo, 2021-11-28:
OK! I just tried this library:

https://pypi.org/project/vcgencmd/

but it also forks a vcgencmd process. Perhaps we could access the vcgen API directly:

https://github.com/raspberrypi/firmware/blob/master/opt/vc/include/interface/vmcs_host/vcgencmd.h

Should we re-implement the vcgencmd python library to avoid the fork calls? It doesn't seem too difficult and doing like this will benefit the whole RBPi user community.

Regards,

jofemodo, 2021-11-28:
BTW, i didn't know that forking a process block all threads, including RT ones. I'm still thinking about it because i think we have several scenarios on zynthian code where things can be improved by knowing this important detail. 

Thanks a lot for opening my eyes to this!!
@riban-bw , what do you think about it? ;-)

jofemodo, 2021-11-28:
Here is the vcgencmd  source:

davidfurey, 2021-11-28:
Thank you for replying so quickly. I'm glad the research is useful. I'm finding it interesting trying exploring quite how low latency the RBPi can go.

> BTW, i didn't know that forking a process block all threads, including RT ones. I'm still thinking about it because i think we have several scenarios on zynthian code where things can be improved by knowing this important detail.

I've struggled to find any documentation that confirms that forking a process blocks all threads, this is just what my investigation suggests - it would be great if someone could confirm that this is expected behaviour.

It seems like implementing the vcgencmd functionality in a library would solve the issue, and be useful to the wider RBPi community.

> I'm still thinking about it because i think we have several scenarios on zynthian code where things can be improved by knowing this important detail.

This is why I was pondering whether it would be good to split the zynthian-ui into two applications (or at least two processes), one that is only responsible for time sensitive audio processing and the other for UI which is less latency sensitive.  I presume there are other system calls, or even patterns of memory access that might cause similar issues.

riban-bw, 2021-11-30:
Hi guys!

Thanks David for your investigation. It is good to see this analysis. I certainly makes sense that running a process from Python and waiting for its result will block. As jofemodo says, we could create a library to access this data or as you suggest, we could obtain the info in a separate thread. Maybe a combination of the two would benefit, i.e. a library that regularly gets the required data running in its own thread, updating a data point / variable that the main UI code can access.

Regarding splitting UI and core functionality - we are working on that right now...

riban-bw, 2021-12-09:
I did a quick and dirty test to disable calling vcgencmd and xruns are reduced / less likely to occur. I think it may be prudent to do this as an interim until we resolve the underlying issue, e.g. implement direct access to vc, e.g. via a lib.

By disabling calls to vcgencmd we lose the indication of under-voltage / over-temperature but it is better to not be warned of a symptom than to have the monitoring of the symptom cause an actual issue.

jofemodo, 2021-12-09:
I don't think this is needed anymore. I moved the vcgencmd call to a separated thread instead of calling it from the main thread. This avoids the main thread getting blocked. This is implemented on the z2rf branch.

riban-bw, 2021-12-09:
@jofemodo  uou marked this ticket with urgent and hot fix tags sure to the impact it had on core operation. I agree which is why I suggested temporarily removing the call to allow users to experience better audio fidelity. It's a simple patch to disable this until we merge the fix in z2rf which I guess will be a while.

jofemodo, 2021-12-13:
The z2rf is now merged on testing. This branch includes the fix i mentioned above:

https://github.com/zynthian/zynthian-ui/commit/eedd72af7bdc6602ce48d0e4976b9cefc6af604b

Sorry, i did it while refactoring the zynthian_gui code, so the change is mixed with a lot of unrelated changes.
This could be a problem for easily merging on stable, specially if we want to "cherrypick". I'm thinking about it ...

Regards,

riban-bw, 2021-12-13:
I tried z2rf and didn't notice substantial difference in xruns so not convinced the change has had much impact. @davidfurey would you have time to perform a similar analysis on the latest _testing_ branch?

davidfurey, 2021-12-18:
I've switched to the latest testing branch, and the pattern has changed slightly.  But fundamentally I still seeing a correlation between xruns and the fork that happens as part of calling `vcgencmd`.  I've verified this using Kernel shark, and by commenting out the call to `vcgencmd` and observing the xruns disappear.

The issue was not the main thread being blocked, so I'm not too surprised that moving the fork off the main thread didn't reduce the number of xruns.

I think we either need to find a way of getting the under-voltage / over-temperature info without forking, or ensure the fork happens in a different process - not just a different thread

[image]
.

steveb, 2021-12-19:
It looks like vcgencmd get_throttled is just reading `/dev/vchiq` and parsing to build the output value. I bet zynthian could read that device file directly for significantly lower overhead. The source for vcgencmd might need to be read to know what to parse
```

root@zynthian-1:~# strace vcgencmd get_throttled
execve("/usr/bin/vcgencmd", ["vcgencmd", "get_throttled"], 0xbeeeecb4 /* 105 vars */) = 0
brk(NULL)                               = 0xd5f000
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6fd9000
access("/etc/ld.so.preload", R_OK)      = 0
openat(AT_FDCWD, "/etc/ld.so.preload", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=54, ...}) = 0
mmap2(NULL, 54, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0xb6fd8000
close(3)                                = 0
readlink("/proc/self/exe", "/opt/vc/bin/vcgencmd", 4096) = 20
openat(AT_FDCWD, "/usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\254\3\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=17708, ...}) = 0
mmap2(NULL, 81964, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f96000
mprotect(0xb6f9a000, 61440, PROT_NONE)  = 0
mmap2(0xb6fa9000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0xb6fa9000
close(3)                                = 0
munmap(0xb6fd8000, 54)                  = 0
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=98423, ...}) = 0
mmap2(NULL, 98423, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f7d000
close(3)                                = 0
openat(AT_FDCWD, "/opt/vc/lib/libvchiq_arm.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0(\26\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=30288, ...}) = 0
mmap2(NULL, 88456, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f67000
mprotect(0xb6f6d000, 61440, PROT_NONE)  = 0
mmap2(0xb6f7c000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5000) = 0xb6f7c000
close(3)                                = 0
openat(AT_FDCWD, "/opt/vc/lib/libvcos.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\210*\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=169364, ...}) = 0
mmap2(NULL, 102288, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f4e000
mprotect(0xb6f57000, 61440, PROT_NONE)  = 0
mmap2(0xb6f66000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8000) = 0xb6f66000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libpthread.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\224O\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=130416, ...}) = 0
mmap2(NULL, 168560, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f24000
mprotect(0xb6f3b000, 61440, PROT_NONE)  = 0
mmap2(0xb6f4a000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16000) = 0xb6f4a000
mmap2(0xb6f4c000, 4720, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f4c000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libdl.so.2", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0 \n\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=9768, ...}) = 0
mmap2(NULL, 73924, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6f11000
mprotect(0xb6f13000, 61440, PROT_NONE)  = 0
mmap2(0xb6f22000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0xb6f22000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/librt.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0 \30\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0644, st_size=26600, ...}) = 0
mmap2(NULL, 90648, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6efa000
mprotect(0xb6f00000, 61440, PROT_NONE)  = 0
mmap2(0xb6f0f000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x5000) = 0xb6f0f000
close(3)                                = 0
openat(AT_FDCWD, "/lib/arm-linux-gnueabihf/libc.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\274x\1\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=1296004, ...}) = 0
mmap2(NULL, 1364764, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6dac000
mprotect(0xb6ee4000, 65536, PROT_NONE)  = 0
mmap2(0xb6ef4000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x138000) = 0xb6ef4000
mmap2(0xb6ef7000, 8988, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6ef7000
close(3)                                = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6fd7000
[... 88 lines cut ...]
Service 2: OPEN (ref 1) 'mmal' remote 81 (msg use 0/3840, slot use 0/15)
  Bulk: tx_pending=0 (size 0), rx_pending=0 (size 0)
  Ctrl: tx_count=10, tx_bytes=500, rx_count=10, rx_bytes=2136
  Bulk: tx_count=0, tx_bytes=0, rx_count=0, rx_bytes=0
  0 quota stalls, 0 slot stalls, 0 bulk stalls, 0 aborted, 0 errors
  instance 7deb57bb
[... cut ...]
  Ctrl: tx_count=8, tx_bytes=412, rx_count=8, rx_bytes=1448
  Bulk: tx_count=0, tx_bytes=0, rx_count=0, rx_bytes=0
  0 quota stalls, 0 slot stalls, 0 bulk stalls, 0 aborted, 0 errors
  instance a8c6fb76
Service 5: OPEN (ref 1) 'mmal' remote 83 (msg use 0/3840, slot use 0/15)
  Bulk: tx_pending=0 (size 0), rx_pending=0 (size 0)
  Ctrl: tx_count=9, tx_bytes=448, rx_count=9, rx_bytes=2044
  Bulk: tx_count=0, tx_bytes=0, rx_count=0, rx_bytes=0
  0 quota stalls, 0 slot stalls, 0 bulk stalls, 0 aborted, 0 errors
  instance 29ae8968
```

steveb, 2021-12-19:
On second thoughts, its not just reading `/dev/vchiq`, its calling ioctls which set up DMA for the reads. This would be hard and risky to do directly from python

steveb, 2022-03-29:
Just coming back to this, it turns out /sys/devices/platform/soc/soc:firmware/get_throttled makes the value available to read, see /sys/devices/platform/soc/soc:firmware/get_throttled

It looks like reading this sysfs path would be a very low overhead alternative to calling vcgencmd

jofemodo, 2022-03-29:
Ouoaaooo!! I wasn't aware of this sysfs interface! Very good point, @steveb !!
Let's try and see if we get free of these annoying XRUNs.

PR #429 (2022-03-31): Read get_throttled from sysfs
This avoids calling vcgencmd, which causes XRUNs in some environments.

Fixes #412
commit 79fba540 (2022-03-31): Read get_throttled from sysfs

This avoids calling vcgencmd, which causes XRUNs in some environments.

Fixes #412
PR #431 (2022-03-31): Read get_throttled from sysfs
This avoids calling vcgencmd, which causes XRUNs in some environments.

Fixes #412
commit 3b81cbb8 (2022-03-31): Read get_throttled from sysfs

This avoids calling vcgencmd, which causes XRUNs in some environments.

Fixes #412
closed 2022-03-31 by jofemodo