zynthian/zynthian-ui #430: XRUN every ~110 seconds with USB audio interface

steveb, 2022-03-30:
I'm attempting to switch from a HiFi Berry audio interface to an external Audient iD4 MKII, it works mostly with these jackd settings:
-P 70 -t 2000 -s -d alsa -d hw:iD4 -r 44100 -p 256 -n 2 --shorts -X raw

However, every 110 seconds or so there is an XRUN. Engine responses vary, pianoteq goes silent and never recovers, zynaddsubfx drops out for about one second then recovers.

Based on the timestamps of 30 events, the average is 111s, minimum 99s, maximum 123s. So it could be something which runs then sleeps for 100s.

According to snoopy, no process is executed at these times, so this isn't like Issue #412

I'm running an up-to-date stable zynthian. I'll continue attempting to diagnose, but I wanted to raise the issue now just in case someone has an idea, or a suggestion for diagnosis approach.

riban-bw, 2022-03-30:
Hi @steveb.

I wonder whether this be best discussed in the Discourse forum rather than issue tracker. There may be more visibility and it doesn't immediately appear to be a Zynthian bug / feature request.

But that said - here are my thoughts...

The Audient iD4 is marketed as a professional soundcard. This often mean that some consumer settings are not included. (Profesisonal = higher cost / fewer features!)  I would start by trying a different samplerate. Try running at 48000 which will do two things:

- Show whether the card is struggling to samplerate operate at 44100
- Show any difference in period of issue occurring which helps diagnose whether related to rate / buffers, etc.

You are preferring 16-bit samples with the `--shorts` option. Try without that and see how that impacts the issue.

Some USB soundcards require 3 buffers. This issue has improved over the years but defo worth trying with `-n 3`.

Again, to see how the issue is affected, try with different buffer sizes. Try 1024 (longer latency) to see if the issue improves / changes behaviour.

Much of this is diagnostics but there _may_ be a solution in there.

Post the results of `lsmod` and `dmesg`. It is interesting / informative to see what drivers are in use and how they are configured.

Have you disabled onboard sound (Headphones)? This is a real xrun fiend.

steveb, 2022-04-04:
I've found a solution, but I'll re-post on Discourse with the fix.

closed 2022-04-04 by steveb