zynthian/zynthian-ui #442: Adding more control screens to pianoteq

rtxanson, 2022-04-30:
Feel free not to prioritize this! I was messing around with the zyngui settings for pianoteq, because I wanted to add some more control screens. I'm noticing that it might be helpful to have the controls learn the current value when the preset / instrument is loaded, rather than supply a default and I'm wondering if this is possible with the way the midi controls are initialized.

I've only been staring at the codebase for a little bit now (and not super familiar with how MIDI works, oops) so it might be this whole question is impossible. Happy to read more code and figure it out too, if you only have time to point me at some lines. 

If it pans out, I can try and turn my aimless hacking into a PR if it's useful for others. Loooove the software so far, thanks all!

riban-bw, 2022-04-30:
Hi @rtxanson. Try using the LV2 version of Pianoteq which exposes most/all parameters.

Please use the project issue tracker: https://github.com/zynthian/zynthian-issue-tracking. We should close these individual ones that have not been used for some time. I'm going to close this ticket. You can check out the LV2 and of you continue to desire a change, post a feature request on the tracker. Cheers!

closed 2022-04-30 by riban-bw

rtxanson, 2022-05-01:
@riban-bw thanks, I'll give LV2 a shot! I missed the general issue tracker, good to know.