zynthian/zynthian-ui #482: Missing 'usb' Python module

oscaracena, 2024-01-05:
I've switched to "chain_manager", updated using webconf and restarted. After that, Zynthian shows an error, failing with the following traceback:

``` 
Traceback (most recent call last):
  File "./zynthian_main.py", line 34, in <module>
    from zyngine.zynthian_chain import *
  File "/home/pi/zynthian-ui/zyngine/__init__.py", line 28, in <module>
    from zyngine.zynthian_engine_fluidsynth import *
  File "/home/pi/zynthian-ui/zyngine/zynthian_engine_fluidsynth.py", line 35, in <module>
    import zynautoconnect
  File "/home/pi/zynthian-ui/zynautoconnect/__init__.py", line 5, in <module>
    from zynautoconnect.zynthian_autoconnect import *
  File "/home/pi/zynthian-ui/zynautoconnect/zynthian_autoconnect.py", line 27, in <module>
    import usb
ModuleNotFoundError: No module named 'usb'
``` 

Installing the Debian package `python3-usb` fixes it.

oscaracena, 2024-01-05:
It also fails to find `alsa_midi`, which can be fixed running `pip3 install alsa-midi`, but I'm not sure if this is the supposed way to do it.

riban-bw, 2024-01-05:
Thanks for repoting these issues. @jofemodo  had noted them and I assumed fixed them but assumptions tend to be poor life decisions... I will add to the install scripts. You are brave ventruing into chain manager - it is bleeding edge development. Expect it to break without warning and your snapshots will not be useable in previous versions (current stable / testing).

oscaracena, 2024-01-05:
> ... You are brave ventruing into chain manager - it is bleeding edge development. Expect it to break without warning and your snapshots will not be useable in previous versions (current stable / testing).

:) Yes, I was aware of that, but I'm working on a controller driver, and I need the most recent version, just to avoid rewriting it later. I like a bit of risk in life ;)

jofemodo, 2024-01-08:
Ups! I also found this issues when updating recent changes in chain_manager, but i missed to include the libraries in the update script. I will do ASAP.

Thanks Oscar!

jofemodo, 2024-01-08:
Done!

closed 2024-01-08 by jofemodo